qmk/qmk_firmware #2367: KC_LGUI and KC_RGUI not triggering cmd on OSX

Nicktho, 2018-02-09:
I'm confused, I've been running qmk on my dz60 rev 2 for a few months now with no problem. Today after booting up OSX KC_LGUI and KC_RGUI stopped working as `cmd` on osx. I'm running El Capitan 10.11.6 (15G1217) which isn't new and this occured without flashing the board. I reflashed the board just in case and still the same issues.

I remapped the cmd keys to space and they work fine, its just KC_LGUI and KC_RGUI not working.

Any ideas?

Nicktho, 2018-02-09:
Also confirmed mapping any other previously working key to KC_LGUI or KC_RGUI does not work.

Nicktho, 2018-02-09:
Using karabiner event viewer shows nothing being sent either.

jackhumbert, 2018-02-09:
With bootmagic enabled, holding down your salt (usually space) + the GUI keys when plugging in the keyboard will disable them. Pressing salt + backspace when plugging it in will reset this!

Nicktho, 2018-02-09:
Yes! Thankyou so much.

closed 2018-02-09 by Nicktho