qmk/qmk_firmware #2384: Add the ability to send shifted keycodes with a mod-tap

dpapavas, 2018-02-12:
I'm opening this issue to draw attention to the pull request (#2055) I've issued a couple of months ago and which I've been using since, without noticing any problems.  I hope this practice is not frowned upon.  An excerpt from the description of the proposed functionality follows:

>This PR introduces a configurable way to use the mod-tap feature to send shifted keystrokes. When enabled, by defining the proper C macro, it treats mod-tap keycodes, which specify shift as part of the modifier list, but where shift is not the only modifier, as if the shift were a part of the keycode. [...] This essentially allows one to use the mod-tap feature to combine shifted keycodes, such as parentheses, with any modifier, except a shift. It gives up the ability to have a mod-tap key send combinations of shift and other modifiers when held, but, since it is configurable, it seems to me like a useful trade-off to have.

drashna, 2018-03-25:
Well, the PR itself should be good enough, I think. 

But just shift isn't enough.  I think that it would need a way to do all of the mods. Either like OSM(), that includes the modifier, or keycodes for each mod.

dpapavas, 2018-03-27:
>Well, the PR itself should be good enough, I think.

 Ok, I'm closing this then and will respond via the PR comments, so that all discussion is gathered in one place.

closed 2018-03-27 by dpapavas