qmk/qmk_firmware #2395: [Feature Suggestion] mod key change layer

JohanJu, 2018-02-13:
Would it be possible/desirable to add a macro ML(mod, layer) it will be like MO(layer) but sending the mod key too. My use-case is that I use Colemak for writing but would like QWERTY for ctrl stuff. 

Here is a quick hack of the functionality for ergodone(ergodox clone)
https://github.com/JohanJu/qmk_firmware/blob/32e9ed2ca102c0bc045f8f3126b22fdb120b4031/tmk_core/common/keyboard.c#L211-L218

optozorax, 2018-03-01:
My solution for this problem:
```C
enum custom_keycodes {
  MY_CTRL = SAFE_RANGE
};

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
  switch (keycode) {
    case MY_CTRL:
      if (record->event.pressed) {
        register_code(KC_LCTRL);
        layer_on(1); // your qwerty layer number
      } else {
        layer_off(1); // your qwerty layer number
        unregister_code(KC_LCTRL);
      }
      return false;
      break;
  }
  return true;
}
```

drashna, 2018-03-25:
You could also do this with `layer_state_set_user`, as well.  

And #2055 is a PR is a proposed change that would introduce this functionality like this

drashna, 2018-03-25:
This was added in #2460, as `LM(mod, Lyaer)`:
https://docs.qmk.fm/feature_advanced_keycodes.html#switching-and-toggling-layers

Closing.

closed 2018-03-25 by drashna