qmk/qmk_firmware #2402: Best code modification to include arduino headers

raycewest, 2018-02-16:
Hello, wanting to integrate some of [this](https://github.com/mrjohnk/ADNS-9800) code into an existing keyboard dir.  What would be the best way to reference outside arduino libraries (e.g., SPI)?

drashna, 2018-02-16:
Not sure if it will help, but a number of the split keyboards use serial communication to link the boards. 
You could probably use that code to do whatever it is that you want to do.

raycewest, 2018-02-16:
Yeah I'm working off of Lets Split.  I'm trying to hack the ADNS-9800 example sketch code into pointing_device code inside of rev2.c.  It's employing SPI, which if I understand correctly, is more specialized than what's found in serial.c,  and it's the only way I've gotten the ADNS to function so far, which is why I'm trying to integrate it.  I think I just have to edit an existing .mk to point to the Arduino libs that only the IDE seems know of.

drashna, 2018-02-16:
Ah, okay.  Well, the dichotemy Keyboard has a mouse/pointer feature that uses serial.  Again, I'm not sure how much help that may be, but ....

Otherwise, @jackhumbert or @fredizzimo would be best able to answer this.

raycewest, 2018-02-16:
Yeah it's honestly more of an Arduino issue rather than a QMK one.  

I've tried appending "-I/usr/share/arduino/..." to CFLAGS in let's split's rules.mk, but "Arduino.h" isn't getting picked up.

raycewest, 2018-02-17:
So, the pin #define's in config_common.h are butting heads with the ones in pins_arduino.h (from the arduino folder).  @jackhumbert, @fredizzimo, could either of you offer any insight on how to integrate functions from an arduino sketch into a qmk keyboard folder?

higherorderfunctor, 2018-03-21:
I replied to your latest issue, but in case anyone else finds this one, I have done it in the fork below to link against FastLED.  The readme.md and bootstrap directory has what you need to include the Arduino headers.  For the promicro you will also need the Sparkfun library.  I haven't played with any other microcontrollers.

https://github.com/kcal-io/qmk_firmware/tree/master/keyboards/9key/keymaps/split_fastled

fredizzimo, 2018-04-08:
I don't think integrating Arduino libraries is an easy task, since they most likely need the whole Arduiono echo system. I'm also not sure if it's even compatible with the current avr-lib and lufa libraries which we are using. 

In that case it has to be added as a completely new platform, in a similar way to ChibiOS.

There are no direct plans, although it has been discussed to integrate Arduino, but if somone is willing to and have time to do it, I think it could make a lot of sense.

jackhumbert, 2020-03-17:
I don't think integrating the raw Arduino headers in a broad sense is something we're going to be able to support - if anything they'd need to be wrappers that reference existing code. But, some individual segments may be able to be integrated and I believe some have.

closed 2020-03-17 by jackhumbert