qmk/qmk_firmware #2411: Tapping Toggle Not Working

bhilburn, 2018-02-18:
Hello!

I'm having a bit of trouble getting Tapping Toggle working in my keymap. I'm not doing anything complex, so I think I must be missing something simple.

Code is here: 
https://github.com/bhilburn/qmk_firmware/blob/tap-toggle/keyboards/dz60/keymaps/cstd60/keymap.c

It doesn't seem to have any effect - it simply continues to function as though the `TT(2)` was an `MO(2)`. I've tried using `OSL` instead, but had the same problem.

Also worth noting is that playing with `TAPPING_TERM` didn't change anything. I tried setting `TAPPING_TOGGLE` to `1` based on [this reddit comment](https://www.reddit.com/r/olkb/comments/7ybtfs/help_tapping_toggle_in_qmk/duf5fkx/), but unfortunately that didn't have any effect.

Ideas?

drashna, 2018-03-25:
Do you have a `TAPPING_TERM` defined?

bhilburn, 2018-03-25:
@drashna - Thanks for the response! Per my original write-up, I did indeed try multiple `TAPPING_TERM` settings:

> Also worth noting is that playing with `TAPPING_TERM` didn't change anything. I tried setting `TAPPING_TOGGLE` to `1` based on this reddit comment, but unfortunately that didn't have any effect.

That said, I was always defining them in my `keymap.c` rather than the `config.h`. I'll give that a go and let you know what happens.

drashna, 2018-03-25:
Ah, I missed that!

And yeah, this needs to be defined in the config.h file or the setting won't be picked up, at all.

drashna, 2018-10-22:
was this still an issue?  

Also, have you tried updating the QMK repo and see if that helps?

mnpqraven, 2018-11-26:
i think the problem lies in the OSL codes or the board themselves. I've been trying to make it work, and the OSM keys are somewhat registering and sending me back single shifted keystrokes but they don't go to hold mode even with the `ONESHOT_TAP_TOOGLE 2` declared in the `config.h`
Here are my [keymap](https://github.com/mnpqraven/qmk_firmware/tree/master/keyboards/dz60/keymaps/othi_60) and [config.h](https://github.com/mnpqraven/qmk_firmware/blob/7df38e1570c9896d0a3f339bea9434b59e5eae3e/keyboards/dz60/config.h#L21-L23)

mnpqraven, 2018-11-26:
an alternative workaround to this is possibly putting OSL mod keys and 'held-down' mod keys to a Tap-Dance key, which i will try to do when i have the time

bhilburn, 2019-01-27:
I realize it's been nearly a year, but I just went back to try some of the suggestions posted in this thread from late last year. It turns out that QMK has changed so much since I first posted this message, that none of my original code works anyway. Closing this out and will re-open if it's still a problem.

closed 2019-01-27 by bhilburn