qmk/qmk_firmware #2412: Another key is executed

baronTommy, 2018-02-19:
A key tap -> Brightness change
Why?

---
- qmk_firmware
  - https://github.com/baronTommy/qmk_firmware

- command
  - `docker run -e keymap=my_keymap -e keyboard=kinesis --rm -v $('pwd'):/qmk:rw edasque/qmk_firmware`

- log
```
QMK Firmware 0.5.236
WARNING:
 Some git sub-modules are out of date or modified, please consider runnning:
 make git-submodule
 You can ignore this warning if you are not compiling any ChibiOS keyboards,
 or if you have modified the ChibiOS libraries yourself.

Making kinesis/stapelberg with keymap my_keymap and target clean

QMK Firmware 0.5.236
WARNING:
 Some git sub-modules are out of date or modified, please consider runnning:
 make git-submodule
 You can ignore this warning if you are not compiling any ChibiOS keyboards,
 or if you have modified the ChibiOS libraries yourself.

Making kinesis/stapelberg with keymap my_keymap

avr-gcc (GCC) 4.8.1
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: ./quantum/matrix.c                                                                       [OK]
Compiling: keyboards/kinesis/kinesis.c                                                              [OK]
Compiling: keyboards/kinesis/stapelberg/stapelberg.c                                                [OK]
Compiling: keyboards/kinesis/keymaps/my_keymap/keymap.c                                             [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/process_keycode/process_leader.c                                                 [OK]
Compiling: ./tmk_core/common/host.c                                                                 [OK]
Compiling: ./tmk_core/common/keyboard.c                                                             [OK]
Compiling: ./tmk_core/common/action.c                                                               [OK]
Compiling: ./tmk_core/common/action_tapping.c                                                       [OK]
Compiling: ./tmk_core/common/action_macro.c                                                         [OK]
Compiling: ./tmk_core/common/action_layer.c                                                         [OK]
Compiling: ./tmk_core/common/action_util.c                                                          [OK]
Compiling: ./tmk_core/common/print.c                                                                [OK]
Compiling: ./tmk_core/common/debug.c                                                                [OK]
Compiling: ./tmk_core/common/util.c                                                                 [OK]
Compiling: ./tmk_core/common/eeconfig.c                                                             [OK]
Compiling: ./tmk_core/common/report.c                                                               [OK]
Compiling: ./tmk_core/common/avr/suspend.c                                                          [OK]
Compiling: ./tmk_core/common/avr/timer.c                                                            [OK]
Compiling: ./tmk_core/common/avr/bootloader.c                                                       [OK]
Assembling: ./tmk_core/common/avr/xprintf.S                                                         [OK]
Compiling: ./tmk_core/common/magic.c                                                                [OK]
Compiling: ./tmk_core/common/mousekey.c                                                             [OK]
Compiling: ./tmk_core/common/command.c                                                              [OK]
Compiling: ./tmk_core/protocol/lufa/lufa.c                                                          [OK]
Compiling: tmk_core/protocol/usb_descriptor.c                                                       [OK]
Compiling: ./tmk_core/protocol/lufa/outputselect.c                                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Class/Common/HIDParser.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c                                        [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c                                [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c                                      [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Host_AVR8.c                                          [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/PipeStream_AVR8.c                                    [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Pipe_AVR8.c                                          [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c                                 [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/ConfigDescriptors.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/DeviceStandardReq.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/Events.c                                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/HostStandardReq.c                                         [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/USBTask.c                                                 [OK]
Linking: .build/kinesis_stapelberg_my_keymap.elf                                                    [OK]
Creating load file for flashing: .build/kinesis_stapelberg_my_keymap.hex                            [OK]
Copying kinesis_stapelberg_my_keymap.hex to qmk_firmware folder                                     [OK]
Checking file size of kinesis_stapelberg_my_keymap.hex                                              [OK]
```

yiancar, 2018-02-19:
I see that you modified the matrix.c file. I would argue that probably something is wrong there

baronTommy, 2018-02-19:
@yanfali 
Is there a cause?
my_keymap is default copy

---
commit log
https://github.com/baronTommy/qmk_firmware/commits/master

commit
https://github.com/baronTommy/qmk_firmware/commit/32a2c579b6e22c02b039f6b1916b7f7d13da27a3

baronTommy, 2018-02-19:
[image]

yiancar, 2018-02-19:
Sorry I dont unserstand. but i think you are using a custom matrx. I dont think you should.

`CUSTOM_MATRIX=yes # need to do our own thing with the matrix
`
disable this

and im pretty sure u dont need this
`+SRC=  matrix.c`

baronTommy, 2018-02-19:
@yanfali
I changed it, but I can not do it well.

https://github.com/baronTommy/qmk_firmware/tree/master/keyboards/kinesis
https://github.com/baronTommy/qmk_firmware/tree/master/keyboards/kinesis/stapelberg

```
# SRC=  matrix.c
CUSTOM_MATRIX=no
```

rules.mk
```

## Project specific files

# SRC=  matrix.c

# MCU name
MCU = at90usb1286
#MCU = atmega32u4

# Processor frequency.
#     This will define a symbol, F_CPU, in all source code files equal to the
#     processor frequency in Hz. You can then use this symbol in your source code to
#     calculate timings. Do NOT tack on a 'UL' at the end, this will be done
#     automatically to create a 32-bit value in your source code.
#
#     This will be an integer division of F_USB below, as it is sourced by
#     F_USB after it has run through any CPU prescalers. Note that this value
#     does not *change* the processor frequency - it should merely be updated to
#     reflect the processor speed set externally so that the code can use accurate
#     software delays.
F_CPU = 16000000

#
# LUFA specific
#
# Target architecture (see library "Board Types" documentation).
ARCH = AVR8

# Input clock frequency.
#     This will define a symbol, F_USB, in all source code files equal to the
#     input clock frequency (before any prescaling is performed) in Hz. This value may
#     differ from F_CPU if prescaling is used on the latter, and is required as the
#     raw input clock is fed directly to the PLL sections of the AVR for high speed
#     clock generation for the USB and other AVR subsections. Do NOT tack on a 'UL'
#     at the end, this will be done automatically to create a 32-bit value in your
#     source code.
#
#     If no clock division is performed on the input clock inside the AVR (via the
#     CPU clock adjust registers or the clock division fuses), this will be equal to F_CPU.
F_USB = $(F_CPU)

# Interrupt driven control endpoint task(+60)
OPT_DEFS += -DINTERRUPT_CONTROL_ENDPOINT

# Boot Section Size in *bytes*
#   Teensy halfKay   512
#   Teensy++ halfKay 2048
#   Atmel DFU loader 4096
#   LUFA bootloader  4096
#   USBaspLoader     2048
OPT_DEFS += -DBOOTLOADER_SIZE=2048

# Build Options
#   change yes to no to disable
#
BOOTMAGIC_ENABLE = yes      # Virtual DIP switch configuration(+1000)
MOUSEKEY_ENABLE = yes       # Mouse keys(+4700)
[... 79 lines cut ...]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/ConfigDescriptors.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/DeviceStandardReq.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/Events.c                                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/HostStandardReq.c                                         [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/USBTask.c                                                 [OK]
Linking: .build/kinesis_stapelberg_my_keymap.elf                                                    [ERRORS]
 |
 | .build/obj_kinesis_stapelberg_my_keymap/common/keyboard.o: In function `keyboard_init':
 | /qmk/./tmk_core/common/keyboard.c:135: undefined reference to `matrix_init'
 | .build/obj_kinesis_stapelberg_my_keymap/common/keyboard.o: In function `keyboard_task':
 | /qmk/./tmk_core/common/keyboard.c:187: undefined reference to `matrix_scan'
 | /qmk/./tmk_core/common/keyboard.c:190: undefined reference to `matrix_get_row'
 | /qmk/./tmk_core/common/keyboard.c:207: undefined reference to `matrix_print'
 | .build/obj_kinesis_stapelberg_my_keymap/common/avr/suspend.o: In function `suspend_wakeup_condition':
 | /qmk/./tmk_core/common/avr/suspend.c:118: undefined reference to `matrix_scan'
 | /qmk/./tmk_core/common/avr/suspend.c:121: undefined reference to `matrix_get_row'
 | collect2: error: ld returned 1 exit status
 |
tmk_core/rules.mk:284: recipe for target '.build/kinesis_stapelberg_my_keymap.elf' failed
make[1]: *** [.build/kinesis_stapelberg_my_keymap.elf] Error 1
Make finished with errors
Makefile:528: recipe for target 'kinesis:my_keymap' failed
make: *** [kinesis:my_keymap] Error 1
```

yanfali, 2018-02-19:
@barronTommy, yanfali != yiancar 😁

baronTommy, 2018-02-19:
@yanfali 
sorry

closed 2018-02-20 by baronTommy