qmk/qmk_firmware #2424: How would I go about making cos-pad firmware?

callumstephenson, 2018-02-21:
I've got no idea how to do anything, would I need drivers and how would I flash QMK etc...

drashna, 2018-02-21:
Start here: 
https://docs.qmk.fm/getting_started_github.html
https://docs.qmk.fm/getting_started_build_tools.html

Fork QMK first.  If you're on Windows, set up MSYS, use the "git clone" option and then run the QMK `util/msys_install.sh` stuff.

That will get QMK set up to "do stuff".

From there: https://docs.qmk.fm/getting_started_make_guide.html

From there, the cospad is already in QMK.  Under `/keyboards/cospad`.  Create a folder under `/keyboards/cospad/keymaps`, ideally with your github username.  So something like `/keyboards/cospad/keymaps/callumstphenson`.   Copy the `keymap.c` from `/keyboards/cospad/keymaps/default`, and start editing that.

Then run `make cospad:callumstphenson` (assuming you used the above folder name), and "magic happens".  If it errors out, it should tell you what line you messed up on.  Fix, and repeat. 

Since I'm guessing it uses DFU for the bootloader... you may be able to run `make cospad:callumstphenson:dfu` to flash the controller, without having to use 3rd party tools.  But worse case, use the QMK Toolbox. It will autodetect everything, except the firmware file.  Just point it at yours (which will be in the QMK firmware folder), and "go".

drashna, 2018-03-25:
Was this still an issue?

drashna, 2018-03-25:
Awesome, glad to hear it then.

Going to close this then.  If needed, reopen it or open a new issue.

closed 2018-03-25 by drashna

szaghi, 2018-11-26:
Hi @callumstephenson @drashna 

I am going to flash my custom keymaps to a cospad, but I was not able to find clear instructions on how flash the qmk compiled firmware on it (I know there should be 2 pins to be shorted in order to enable flashing, but I do not know which pins and how to short them).

Can you help me?

Thank you very much.

szaghi, 2018-11-26:
Never mind, I have found the guide on kprepublic, thanks you anyway!

Cheers