qmk/qmk_firmware #2425: ps2avrGB build fails

knanan, 2018-02-21:
the result of make is:

```
907a8deb:qmk m$ make ps2avrGB VERBOSE=1
# Check if we have the CMP tool installed
cmp .//Makefile .//Makefile >/dev/null 2>&1; if [ $? -gt 0 ]; then printf "\033[31;01mError:\033[0m\033[1m cmp command not found, please install diffutils\n\033[0m"; exit 1; fi;
# Check if the submodules are dirty, and display a warning if they are
if [ ! -e lib/chibios ]; then git submodule sync lib/chibios && git submodule update --init lib/chibios; fi
if [ ! -e lib/chibios-contrib ]; then git submodule sync lib/chibios-contrib && git submodule update --init lib/chibios-contrib; fi
if [ ! -e lib/ugfx ]; then git submodule sync lib/ugfx && git submodule update --init lib/ugfx; fi
git submodule status --recursive 2>/dev/null | \
	while IFS= read -r x; do \
		case "$x" in \
			\ *) ;; \
			*) printf "\033[33;01mWARNING:\033[0m\n Some git sub-modules are out of date or modified, please consider runnning:\033[1m\n make git-submodule\n You can ignore this warning if you are not compiling any ChibiOS keyboards,\n or if you have modified the ChibiOS libraries yourself. \n\n\033[0m";break;; \
		esac \
	done
rm -f .//.build/error_occurred > /dev/null 2>&1
# Run all the commands in the same shell, notice the + at the first line
# it has to be there to allow parallel execution of the submake
# This always tries to compile everything, even if error occurs in the middle
# But we return the error code at the end, to trigger travis failures
if [ -f .//.build/error_occurred ]; then printf "\033[31;01mMake finished with errors\n\033[0m" & exit 1; fi;
if [ -f .//.build/error_occurred ]; then printf "\033[31;01mMake finished with errors\n\033[0m" & exit 1; fi;
```

knanan, 2018-02-21:
All the special prerequisites are installed, other keyboard builds are `[OK]` - this one just fails silently without verbose

drashna, 2018-02-21:
Yeah, can confirm that this is happening.   And it's not a chibiOS issue, as I tested other boards, and they compile file.

ledoge, 2018-02-27:
I did a `git bisect`, it's caused by commit 7b754e1a5a902abd51ad314c68ef1606d89adf95.

garrettmurray, 2018-03-05:
Also having this problem which is preventing attempting to convert my Pearl from bootmapper to QMK.

garrettmurray, 2018-03-05:
@Sebb767 Any ideas?

Sebb767, 2018-03-05:
The problem seems to be that there's no 'default' keyboard for ps2avrGB anymore. 

Initially, it only supported the b.mini (one of the winkeyless.kr boards), but as @jetpacktuxedo and I added support for the b.fake and b.face we moved those to subprojects (those are now standalone folders).

I'm pretty sure I could fix it by adding a default keymap again, but I'm not sure how much sense this  would make as it is more of a firmware rather than a specific keyboard. If you want I can do a PR, though.

garrettmurray, 2018-03-05:
@Sebb767 If you wouldn't mind doing a PR, that'd be great. Then I'll patch it and see if I can make again.

Sebb767, 2018-03-05:
@garrettmurray PR #2478

garrettmurray, 2018-03-05:
Thanks, Sebb. That worked perfectly. Left a review on the PR.

PR #2478 (2018-03-07): Fixed compilation of the ps2avrGB keyboard/firmware (resolving #2425)
This PR fixes the silent compilation error for the ps2avrGB keyboard/firmware. This error was caused by a lacking default keymap which it did not have because all keyboards based on it were moved to another directory. I also added the required config.h options so it's possible to compile it again and (probably) flash it on a b.mini.

Lastly, I updated the README to reflect the current state.

This commit fixes #2425
commit bec8d58a (2018-03-07): Fixed compilation of the ps2avrGB keyboard/firmware (resolving #2425) (#2478)

* Fixed compilation of the ps2avrGB keyboard/firmware

This commit fixes the silent compilation error for the ps2avrGB
keyboard/firmware. This error was caused by a lacking default
keymap which it did not have because all keyboards based on it were
moved to another directory. I also added the required config.h
options so it's possible to compile it again and (probably)
flash it on a b.mini.

Lastly, I updated the README to reflect the current state.

This commit fixes #2425

* Referenced the pearl in the ps2avrGB RE [... cut]
closed 2018-03-07 by jackhumbert