qmk/qmk_firmware #2439: Can't chain LGUI(LCTL(LSFT(KC_4)))

thinkv, 2018-02-26:
Trying to create a screenshot key in mac however the chain "LGUI(LCTL(LSFT(KC_4)))" seems to work once after flashing and then never again. After the first time it simply returns a "4".

Any ideas?

TurboMech, 2018-02-26:
I just created a shortcut inside of System Preferences > Keyboard > Shortcuts, then inputted whatever that key combination was into QMK.

thinkv, 2018-02-26:
@TurboMech but what key combination is not taken and available to QMK? I thought Command Control Shift 4 was already a shortcut

drashna, 2018-02-26:
Tried this on my ergodox, no issues with it.  But on windows.

What keyboard are you using?

thinkv, 2018-02-26:
I'm using a Clueboard, rev3 pcb

here is my keymap file https://gist.github.com/thinkv/105a5b33057f6d72f3fd17a992a97011

TurboMech, 2018-02-26:
@thinkv I use `LCTL(KC_S)` and have that as a shortcut so in Mac it's just CTRL and S.  That is my keypress for my F5 key on my 75%.

TurboMech, 2018-02-26:
Then make it something crazy like LCTRL(LALT(LGUI(S))) then change the shortcut in Mac

thinkv, 2018-02-26:
That's what I have now and it's not working as per this issue @TurboMech

thinkv, 2018-02-27:
I'm stupid, I changed the mapping on OSX and it messed things up, it did play funny though and clearly the mapping doesn't work at the level with OSX. Sorry all.

closed 2018-02-27 by thinkv