qmk/qmk_firmware #2440: CTL_T(KC_LPRN) Does not work as expected

Noia, 2018-02-26:
My understanding is that CTL_T(KC_LPRN) should
  When tapped: insert ( which is SHIFT + 9.
  When held: hold left control.

In actuallity this appears to insert a KC_9 when tapped, and hold MOD_LCTL | MOD_LSFT when held.
Tried this on an Iris and Mechmini on linux.

drashna, 2018-02-26:
I'm not sure there is a good solution here.  The KC_LPRN isn't it's own keycode, but is a placeholder for a pre-shifted 9.  
https://github.com/qmk/qmk_firmware/blob/master/quantum/quantum_keycodes.h#L495

Aside from that, not sure there is a solution or how much help I could be.

Noia, 2018-02-26:
The issue here is that the Shift modifier is applies incorrectly. It should be either Ctrl or Shift+9, why is it giving me Ctrl+Shift or 9? Somehow the shift is migrating from the momentary part to the hold part of MT function.

drashna, 2018-02-26:
I think that the shift is getting stuck, because of how the keycode is being handled.

fauxpark, 2018-02-26:
It looks like this is not really possible due to the way that keycodes are stored in QMK.

- The 'keycode" `CTL_T(KC_LPRN)` resolves to `0x6326`.
- `0x6000` represents a [mod-tap event](https://github.com/qmk/qmk_firmware/blob/master/quantum/quantum_keycodes.h#L80).
- [The second digit is the modifiers.](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/common/action_code.h#L212) `0x01` is the Ctrl from the CTL_T macro, `0x02` is the Shift from the LPRN keycode. Interestingly the left/right flag is actually the 1 bit of the previous digit, so in fact specifying a right mod will cause all the other mods specified to become right mods.
- `0x26` is of course the keycode for `KC_9`.

Because modifiers are only assigned a single portion of the keycode, the Ctrl and Shift are merged. The mod-tap feature simply tells QMK to switch between the mod portion and the "actual" keycode.

oscar-b, 2018-03-01:
Also see https://github.com/qmk/qmk_firmware/pull/2055

closed 2018-10-22 by drashna