qmk/qmk_firmware #2445: Errors when building for atmega328p

Maartenwut, 2018-02-27:
I have a board with an atmega328p which I'm trying to port to QMK, but I'm getting errors when building. You can find my code here: https://github.com/Maartenwut/ydpm40-qmk.

The error I'm getting is:

> Compiling: ./tmk_core/common/avr/bootloader.c                                                      ./tmk_core/common/avr/bootloader.c: In function ‘bootloader_jump’:
> ./tmk_core/common/avr/bootloader.c:156:13: error: ‘UDCON’ undeclared (first use in this function)
>              UDCON = 1;
>              ^
> ./tmk_core/common/avr/bootloader.c:156:13: note: each undeclared identifier is reported only once for each function it appears in
> ./tmk_core/common/avr/bootloader.c:157:13: error: ‘USBCON’ undeclared (first use in this function)
>              USBCON = (1<<FRZCLK);  // disable USB
>              ^
> ./tmk_core/common/avr/bootloader.c:157:26: error: ‘FRZCLK’ undeclared (first use in this function)
>              USBCON = (1<<FRZCLK);  // disable USB
>                           ^
> ./tmk_core/common/avr/bootloader.c:158:13: error: ‘UCSR1B’ undeclared (first use in this function)
>              UCSR1B = 0;
>              ^
>

fredizzimo, 2018-04-08:
You probably need a newer version of the AVR compiler. If the package manager of your operating system doesn't provide you one, you can install it from here http://www.microchip.com/avr-support/avr-and-arm-toolchains-(c-compilers)

drashna, 2018-05-11:
Was this still an issue, @Maartenwut ?

Maartenwut, 2018-05-12:
I'll have another go at it tonight.

Maartenwut, 2018-05-15:
Still an issue unfortunately, same error. Even updated avr-gcc to 5.4.

drashna, 2018-05-15:
What OS is this?

Maartenwut, 2018-05-15:
Debian Sid on a virtual machine. Are you able to build successfully?

fauxpark, 2018-05-16:
These errors are occurring because the symbols `UDCON`, `USBCON`, `FRZCLK` and `UCSR1B` don't exist in the [328P-specific](http://svn.savannah.gnu.org/viewvc/avr-libc/trunk/avr-libc/include/avr/iom328p.h?revision=2460&view=markup) `io.h` ([32U4 for comparison](http://svn.savannah.gnu.org/viewvc/avr-libc/trunk/avr-libc/include/avr/iom32u4.h?revision=2536&view=markup)). And it is not that surprising - the 328P does not have any inbuilt USB functionality like the 32U4 does, so those registers don't physically exist.

Try changing [this line](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/common/avr/bootloader.c#L154) to:

```c
#ifndef __AVR_ATmega328P__
```

The 32A also does not have USB, so the block is ignored if we're compiling for that chip.

drashna, 2018-05-16:
@fauxpark If that works for them, wouldn't it be a good idea to change that line to:
```c
#if !defined(__AVR_ATmega32A__ ) && !defined(__AVR_ATmega328P__)
```

fauxpark, 2018-05-16:
It would, I just didn't recall OTOH exactly how to chain multiple `ifndef`s - that looks about right.

Maartenwut, 2018-05-16:
Works like a charm! Thanks guys 👍 
Who's gonna do a pull request to edit this line?

drashna, 2018-05-16:
ooo ooo, you are! 

:)

Nah, I'll do it

closed 2018-05-21 by skullydazed

RivasMario, 2023-08-22:
if anyone finds this, for atmega328p you need to use usbaspoader bootloader

PR #2986 (2018-05-21): Fix compile issues for atmega328p
This will fix #2445