qmk/qmk_firmware #2447: Keyboard name limitation?

DaiUrakawa, 2018-02-28:
When I make a project named "test" with new_project.sh and try to do `make test:default`, it just prints qmk version and quits.
There seems to be limitation in keyboard name. Happy if this kind of thing is documented [here.](https://docs.qmk.fm/hardware_keyboard_guidelines.html#naming-your-keyboardproject)

loadedsith, 2018-03-01:
I imagine that this is related to `make test` being a reserved word of some sort. I've created a pull request that adds the documentation for `test`,`all`, and `keyboard`. Does anyone know if there are others that should be added?

skullydazed, 2018-03-03:
I think that covers it. Thanks for the helping hand @loadedsith

closed 2018-03-03 by skullydazed