qmk/qmk_firmware #2461: Unicode Input

Sinderella, 2018-03-02:
My idea is to be able to type £ sign with US layout and/or to be able to use US layout on the UK layout computer. Before I am trying to achieve the later (not even sure it's achievable), I am stuck with inserting the pound sign, unicode.

I have `UNICODE_ENABLE = yes` and I use `UC(0x00A3)` in the keymap, but once I tried to press that key, it prints out `00a3` instead of the pound sign. (I got the code from http://graphemica.com/%C2%A3). Is there anything I've done incorrectly?

I've tried on both macOS and Windows 10, I only need it to work on Windows, for my work laptop.

Cheers

Sinderella, 2018-03-02:
https://docs.qmk.fm/how_keyboards_work.html explains why I cannot achieve it

closed 2018-03-02 by Sinderella

niico100, 2022-09-02:
I use AutoHotKey for this