qmk/qmk_firmware #2465: Two same-mod Mod-Taps: Make behavior pressing both configurable

Nebukadneza, 2018-03-02:
Hi All,

first of all, thanks for this great firmware. It really makes keyboard-tinkering a fun endavour :+1: .

One thing i stumbled upon when trying to map my favorite layout¹ onto an ergodox was a strange behavior in the Mod-Tap functionality. I’m using Mod-Tap on 2 home-row keys, to switch neo-layout layers (using ALGR and GUI as modifiers). After enabling IGNORE_MOD_TAP_INTERRUPT this works like a charm. However, the problem arises in the common case of holding the ALGR mod on the right side, and using the now-utility-layer on the left. Now on the left, there’s a cursor key „under“ the KC-part of the Mod-Tap key. So … to be able to quickly scroll around I’d like to be able to hold that key. However, when holding, the Mod-hold triggers again.

I know this is kind of a … non-problem since it’s exactly doing what i want. But I was hoping for some ideas or guidance on how to maybe resolve this. One idea could be to be able to specify a blacklist of keys which, when a certain MT is pressed, are never considered to be held …?

I also realize this is quite the peculiar problem. However, I could not solve it using simple macros in my keymap … in fact, i could not even get Mod-KC-dualroling to work as smooth as with the builtin macros. It’d always lag a tad more, and never be as „roll-over~y“ as IGNORE_MOD_TAP_INTERRUPT + MT.

I’d be really glad for any kinds of feedback or help ….

Best & Thanks,
- Dario

¹: http://neo-layout.org/

skullydazed, 2018-03-03:
I'm not entirely sure I understand your problem correctly, but I think I do. There are a couple ways to approach this (and others can chime in if they have a more elegant solution to this problem.)

The first is to use `process_record_user()` to catch this situation and send the appropriate key codes instead. I'd need to do more work than I have time for right now to put this solution together, but if you're up to noodling through the code yourself this would be the best starting place: https://docs.qmk.fm/custom_quantum_functions.html

The other possibility would be to define a second layer that wraps your primary layer in `RALT()`. I do a similar thing to make a "combined" shift/fn key here: https://github.com/qmk/qmk_firmware/blob/master/keyboards/clueboard/66/keymaps/shift_fn/keymap.c

There are drawbacks, because you have to basically define your keymap twice, and make changes in both places, but that would allow you to have the cursor key there while holding altgr instead of the mod-tap key. You also need to be aware that you won't be able to hold altgr and have your computer see that, so if you need to e.g. hold it while clicking on something you'll need to put another altgr somewhere. Some people get around that by assigning altgr to the spacebar on the layer their altgr key activates.

Nebukadneza, 2018-03-04:
Hi,

thank you for your quick answer, and suggestions. Indeed, the copy’n’wasted layer which is `RALT(…)`ed  throughout didn’t occur to me. While it’s ugly to modify, it is easy to generate with a bit of sed magic, and perfectly serves my purpose!

Thanks a lot!
-Dario

closed 2018-03-04 by Nebukadneza