qmk/qmk_firmware #2470: New way to toggle layer request

lbibass, 2018-03-03:


pattulus, 2018-03-03:
I can see this would be a nice thing to have, but I can also see unwanted behaviors, when for instance only parsing the last 10 entered characters.

I set up the leader key to have a similar behavior:

      //Layer switching
      SEQ_ONE_KEY(KC_1) { ang_tap (TT(NUMPAD), 0); } // Numpad
      SEQ_ONE_KEY(KC_2) { ang_tap (TT(MOUSE), 0); } // Mouse

Of course you could just use `SEQ_FOUR_KEYS` to have a longer input sequence.

lbibass, 2018-03-03:
@pattulus well of course there would be some limitations, like the ten entered characters. Most layer names are simple after all. Thanks for the help! So basically my idea basically already exists in leader key?

pattulus, 2018-03-04:
If you're willing to accept the leader key as _the first letter to enter_ in the word that represents your layer, then yes.

lbibass, 2018-03-04:
could it also be more than one letter if I used something like SEQ_THREE_KEY?

pattulus, 2018-03-04:
Yep, @jackhumbert implemented up sequences up to 5 letters on [June, 22](https://github.com/qmk/qmk_firmware/commit/98f0807359cfa78d25442b91ff4c5bbfc5679661).

closed 2018-10-22 by drashna