qmk/qmk_firmware #2475: [lets split] RGB light and layer_state_set does not work together.

bbaserdem, 2018-03-05:
From what I gather, `layer_state_set_user` function is executed at layer change, passing the new layer configuration with the variable `uint32_t state`

This is the behaviour I get on the planck board, and use this function to change RGB indicators. When the same code is used on a pro micro (let's split) the function executes with the state that I'm switching from, instead of the state I am switching to. So my RGB change function lags behind a layer. If I activate `layer 1` on top of `layer 0`, it looks like instead of getting `0x00000003` I get `0x00000001`, and when I deactivate `layer 1` it executes as if `0x00000003` is passed as an argument.

Note that the code that uses this functionality comes from my userspace, so I can verify that the same code works differently on two boards. My code can be found [here](https://github.com/bbaserdem/qmk_firmware/blob/master/users/bbaserdem/bbaserdem.chttps://github.com/bbaserdem/qmk_firmware/blob/master/users/bbaserdem/bbaserdem.c#597). I tested this numerous times and can confirm that it works on a planck (rev4) and not a lets split.

All the layers are working fine, the keypresses are sent correctly.

drashna, 2018-03-05:
`rgblight_enable();` isn't needed except for in the `matrix_init_*` function.  I mean, it shouldn't hurt, but ... it's unnecessary. 

Why are you doing this: `state = layer_state_set_keymap (state);`. rather than ` return layer_state_set_keymap (state);`?
The reason I ask, is because I used the return option explicitly because it was step up that way in the core files: 
https://github.com/qmk/qmk_firmware/blob/4ea3bbdb4cbb02310623c9dfc464d84e4f653492/tmk_core/common/action_layer.c#L71-L74

Also the _keymap functions in the _user should be the last thing called, usually.  you're calling it first in the matrix_init_user function, for example. 

And it looks like you were trying to save the base layer?  If you're not anymore, then moving all the RGB stuff into the `layer_state_set_` function, as it doesn't appear to be called anywhere else. 

Also, if you're sharing keymaps .... you should use the layout feature. 
https://docs.qmk.fm/feature_layouts.html

(sorry if I'm missing anything, as I'm a bit tired, atm)

bbaserdem, 2018-03-05:
I tried to add a delay specific to the keymap before execution of the loop. That is an artifact of my many trials to fix this issue. I can confirm that fixing it doesn't help.

The rgblight_enable is due to indicator lights only being on outside the default layer. I usually configure it to off to reduce power drain.

drashna, 2018-03-05:
Ah, well, I can see about writing something else that would work.

As for power consumption, you may want to add a define for "RGBLIGHT_LIMIT_VAL".  This will limit the brightness, basically.  So, you could limit the power draw

drashna, 2018-03-25:
Were you able to fix this?

closed 2018-05-11 by drashna