qmk/qmk_firmware #2477: Increase the time required for a hold without affecting TAP_DANCE

danielo515, 2018-03-05:
Hello.

Currently I use space both as space and as shift keys. However, when I type fast I am facing a big problem: sometimes I want an space and I get a shifted key (ej: helloBuddy instead of hello buddy).

I can increase TAPPING_TERM, but that will affect tap dance, which uses (badly chosen IMO) the same config variable.

Any advice will be greatly appreciated.
Regards

danielo515, 2018-03-07:
I have a solution that almost works. It is not my code, it was suggested by a user at gitter, I think his name is @fredizzimo .
The only non working part from his code is effectively determining the pressed keycode. I'll document here the changes:

add the following to `action_tapping.c`

```c
// This was the old line
// #define WITHIN_TAPPING_TERM(e)  (TIMER_DIFF_16(e.time, tapping_key.event.time) < TAPPING_TERM)
#define WITHIN_TAPPING_TERM(e)  (TIMER_DIFF_16(e.time, tapping_key.event.time) < get_tapping_term(&e))
// Overridable tapping term
__attribute__ ((weak))
uint16_t get_tapping_term(keyevent_t* event) {
    return TAPPING_TERM;
}
```

Of course add the declaration to `action_tapping.h`

```c
uint16_t get_tapping_term(keyevent_t* event);
```

Then you can override the `get_tapping_term` function at any point on your keymap:

```c
uint16_t get_tapping_term(keyevent_t* event) {    
    uint16_t keycode = keymap_key_to_keycode(layer_switch_get_layer(event->key), event->key);
    if (keycode == SFT_T(KC_SPC) )    {
        return TAPPING_TERM + 1250;
    }
    //dprintf("Keycode found is %u", keycode);
    return TAPPING_TERM;
}
```

However the keycode identification thing is not working. I know that it works because if I omit the if statement and just send an high time it works as expected. But it is kinda useless without identifiying the key pressed.

drashna, 2018-05-11:
Why couldn't this use a case?  Or could it?

drashna, 2018-05-11:
You mean that it doesn't work for `SFT_T(KC_SPC)`, but only `SFT_T`?

drashna, 2019-01-30:
Did you ever get this working?

drashna, 2019-01-31:
Well, if you didn't, I did.

Check out #5009. I tracked down the issue here, and got this working.  And the solution/problem is in the comments.

danielo515, 2019-01-31:
I'll check it out! Thanks

jackhumbert, 2020-03-17:
Now that #5009 is merged, I think this can be closed - if you have any issues with implemenation, feel free to comment here!

closed 2020-03-17 by jackhumbert