qmk/qmk_firmware #2492: what is now the correct way to build community keymaps

gid0, 2018-03-07:
The folder layout is changing really much, quite often.
What is now the correct way to build a community keymap for a particular keyboard ?

Let's say I want the bepo keymap for an Ergodox keyboard. I used to issue :
`make KEYBOARD=ergodox KEYMAP=bepo`
Given the examples and readme, I can't find a way to compile it.

lindhe, 2018-03-08:
I can't figure it out either. Trying to make `layouts/community/ergodox/swedish-lindhe`, but can not figure it out. Have tried `make ergodox:swedish-lindhe` as well as `make keyboard=ergodox keymap=community/swedish-lindhe` but neither works.

gid0, 2018-03-08:
I think I finally got it, but I'm glad to see I'm not the only one who struggled. Try replacing ergodox with ergodox_ez :
`make KEYBOARD=ergodox_ez KEYMAP=swedish-lindhe`
or
`make ergodox_ez:swedish-lindhe`
Somewhat counter-intuitive when you don't have an Ergodox "EZ".

lindhe, 2018-03-09:
> https://docs.qmk.fm/getting_started_make_guide.html
>
> For the ergodox, you want to use the ergodox_ez keyboard. It's the same
> design, so it was folded into one.
>
> But for bepo, you'd run make ergodox_ez:bepo (or optionally make
> ergodox_ez:bepo:teensy)

Sounds like a strange decision. Ergodox is the original design; EZ is the
variation.

drashna, 2018-03-09:
It kind of does.  But I suspect that the reasoning for that is that ezuk maintains the code for the ergodox And sihce he is And since he's the creator of the EZ, and dominic (the ergodox creator), well doesn't maintain the code here (not that I've seen).... that it makes more sense to handle it that way.

But since the boards are identically designed, it shouldn't be an issue, firmware wise.

lindhe, 2018-03-09:
Ah. What further complicated matters was that my fork was based on a broken commit. It works.

drashna, 2018-03-25:
Oh no! :(

Well, merging/rebasing may help?  Otherwise, grab your keymap files, blow it away, and restart.

closed 2018-05-11 by drashna