qmk/qmk_firmware #2495: KC_LOCK not working for OSM modifiers

danielo515, 2018-03-08:
Hello,
 i'm trying to use KC_LOCK on my keymap and I want to use it with my one shot modifiers. The documentation of KC_LOCK says that this should be possible, but it is not working. Here is how I have my modifiers defined:

```
#define KC_ALT OSM(MOD_LALT)
#define KC_CTL OSM(MOD_LCTL)
#define KC_SFT OSM(MOD_LSFT)
```

It is working for normal keys (for example letters and numbers).

drashna, 2018-03-08:
Why not define these: 
```
#define ONESHOT_TAP_TOGGLE 5  /* Tapping this number of times holds the key until tapped this number of times again. */
#define ONESHOT_TIMEOUT 5000  /* Time (in ms) before the one shot key is released */
```

Then they'll lock by tapping the correct number of times? 

And as for why, I think it's how the lock key works.  I suspect it only works with normal keys?

danielo515, 2018-03-10:
Hello @drashna, thank you for your answer.
I also thought about that as an alternative solution, however if possible I prefer to just hit two keys rather than hitting the same key 5 times for activating and another 5 times for deactivating. Of course I can lower the number, but if I lower it too much it may become too easy to hit it by accident. 
You say that it may not work because it's nature, but the documentation clearly says that it should work for OS modifiers, so or documentation is wrong or implementation is wrong.

Regards

squarology, 2018-03-16:
I'm also having trouble getting this to work. The documentation clearly states that this is possible. So should the documentation be changed?

Or is this a bug?

Thank you.

drashna, 2018-03-25:
@danielo515 That's why you change the tap toggle to 2.  Once for the next key only, double tap to lock. 

And I think that the KC_LOCK doesn't work with "function" keys like OSM

danielo515, 2018-03-26:
Hello @drashna 
Tap toggle to 2 stills to low. However, what I have discovered (contrary to what documentation says) is that you have to tap the key as many times as ONESHOT_TAP_TOGGLE defines, but you only have to tap it once to deactivate. For that reason, a configuration of ONESHOT_TAP_TOGGLE of 3 is more or less as good as using KC_LOCK

drashna, 2018-03-26:
I've found that it works that way too (one click to turn it off).  But I think that's intentional, actually.... just not well documented.  

But I'm glad to hear that you did find a config that works for you!