qmk/qmk_firmware #2498: Automatic glossary links in docs are overzealous

henrebotha, 2018-03-08:
For example, the letter "C" on [this page](https://docs.qmk.fm/feature_leader_key.html) is automatically hyperlinked to the glossary definition for the C language, **even though it literally just means the <kbd>C</kbd> in this context**. There are many similar examples throughout the docs — the word "make" is frequently used to mean things _other_ than the build tool, yet it hyperlinks to the glossary every single time. 

This hurts legibility and potentially confuses the reader. Is there a way to remove this feature?

skullydazed, 2018-03-08:
I've been thinking this myself lately. Seems like gitbook doesn't have a way to do that, but we can rename it to `qmk_glossary.md` and update the links.

henrebotha, 2018-03-08:
What would renaming the file achieve? Would it stop the hyperlinking or something?

skullydazed, 2018-03-08:
Yes, from gitbook's perspective we have a glossary because a file called glossary.md exists in the right format. By renaming it gitbook won't think we have a glossary.

commit 56639979 (2018-03-08): Disable the glossary linking to fix #2498 (#2499)
closed 2018-03-08 by skullydazed