qmk/qmk_firmware #2514: Leader Not Working

shmcgrath, 2018-03-10:
My/the leader key is not working. When I hit the leader key followed by the sequence, I get the character I am typing. I have tried it with 1, 2, and 3 keys in sequence. I have made sure that the key is not physically broken. The code is compiling fine and loads onto my Planck without issue. [Here is they keymap I am using.](https://github.com/shmcgrath/qmk_firmware/tree/master/keyboards/planck/keymaps/shmcg) Let me know if you need further information.

jackhumbert, 2018-03-10:
Here are some other threads on /r/olkb that were having issues as well:

* https://www.reddit.com/r/olkb/comments/837dyv/leader_key_troubles/
* https://www.reddit.com/r/olkb/comments/7u7362/leader_key_timing/

randywallace, 2018-04-27:
AH HA.  I was tracking this one down, and put together that those settings _must_ go in config.h for them to take effect; otherwise, the default LEADER_TIMEOUT (200ms), which is _very_ short, is set. 

https://docs.qmk.fm/reference/config-options

closed 2018-05-16 by jackhumbert

drashna, 2018-12-15:
Actually, I think upping the default timeout would be a good idea too.

PR #2818 (2018-05-16): Add default to LEADER_TIMEOUT in case it's not defined
This should fix #2514, in theory.

`LEADER_TIMEOUT` has no default value defined. So it it's not defined, it won't trigger properly.