qmk/qmk_firmware #2527: Caps lock LED as a normal backlight controlled LED

TurboMech, 2018-03-14:
Hey all, I can't for the life of me figure out how to make the LED indicator for caps lock to operate as a normal LED that is controllable, instead of it acting as the indicator led.  This is for an ALU84 but also applies to other keyboards that have caps lock indicator LEDs built in.

Thanks for any input

drashna, 2018-03-14:
https://github.com/qmk/qmk_firmware/blob/master/keyboards/alu84/keymaps/default/keymap.c#L86-L91

drashna, 2018-03-14:
Specifically, I think you'd want: 

```
void matrix_init_user(void) {
  DDRB &= ~(1 << 2); 
  PORTB &= ~(1 << 2);
}

void caps_led_enable(void) {
  DDRB &= ~(1 << 2); 
}
void caps_led_disable(void) {
  DDRB |= (1 << 2); 
}
```

But I could be wrong about that

TurboMech, 2018-03-14:
@drashna no I'm trying to get it to not act as a indicator, just a normal backlight LED

drashna, 2018-03-14:
You mean to be used with the rest of the backlighting?  

No, I don't think so, since it's running on a completely separate pin than the rest of the backlighting.

WarmCatUK, 2018-03-17:
Could you have it copy the state of the regular backlights? Just a thought.

drashna, 2018-03-25:
Did you get this resolved?

TurboMech, 2018-03-25:
I did not.  I spent a few days working on it to no solution.  Best I can think is that the LED indicator is mostly hardwired to the MCU and therefore cannot be manipulated.

drashna, 2018-03-27:
If it's hardwired, then the above (what I linked) should still work. 

Stripe the DDR/PORT, and `(1 <<` out of the code, and you get the pin (B2, in this case).

Taudris, 2018-03-29:
If I read right, you're trying to make the capslock key be lit up the same way as the rest of the keyboard.

Assuming that's true, there is no good software solution possible. The backlighting is driven by pin `B6` and is dimmed using a timer output. The LED for the capslock light is on pin `B2`, which does not have a timer output (see the [ATmega32U4 datasheet](https://www.pjrc.com/teensy/atmega32u4.pdf) page 72 section 10.3.2), and thus cannot be dimmed (at least not without significant flickering).

The best solution is a hardware modification: cut the PCB trace connecting pin `B2` to the capslock key's LED and connect it to one of the adjacent LEDs by soldering a small wire. You would of course lose the option of having a capslock indicator by doing this.

closed 2018-10-22 by drashna