qmk/qmk_firmware #2536: YMD96 Atmega32A led backlight issue

harshitgoel96, 2018-03-15:
I have Ymd96 running QMK from the default branch. The keyboard itself is working fine but backlight is not working. I have using following settings on QMK

BACKLIGHT_PIN D4 (Confirmed with Vincent on pin as well)
BACKLIGHT_LEVELS 3

Tried turning on leds using

BL_TOGG
BL_ON/BL_OFF

The Caps Lock, Scroll Lock, and Num lock do not work.

drashna, 2018-03-25:
Make sure that you've updated the repo.

And check out krusli's keymap files, as those "should" work.

harshitgoel96, 2018-03-26:
I hope you meant the merge he created for jj40/50

i have used that repo but not the keymap file, i basically used his JJ40 code with my custom keymap.... and that includes some funny issue. 

1) LED is still not working.
2) the keyboard got stuck with caps lock pressed on.

Would using the exact keymap solve the issue above ? and do you recommend that i should use the krusli's hex on my board directly ?

drashna, 2018-03-26:
Well, the jj40 looks to be using a custom backlight implementation, as well as custom RGB driver.  That may be why. 

You can see this in the custom rgblight driver, as well as the backlight.c source included in the board (both referenced in the rules.mk for the board).  So that may be a good place to start/look into to get this working.

harshitgoel96, 2018-03-26:
i did not notice that, and I did try using that code... Maybe my approach was wrong. Will check and update on it later, and may be add a pull request as well.

drashna, 2018-03-26:
well, glad that I mentioned that, then!  And hopefully, that will help!

harshitgoel96, 2018-03-26:
@drashna  making ymd96 after modifying the jj40 krusli works the right way for backlight and rgb. There is one small issue however. Caps lock gets stuck. 

I think this is due to some minor difference between jj40 and ymd96.

when i press capslock on any keyboard attached to the computer where ymd96 is attached, the keyboards stops working, switch hitter reports that caps is stuck on down press. 

there is one more issue related to this, when i press the num lock key, it enables to wrong key, which i think is causing the issue.

harshitgoel96, 2018-03-26:
@krusli your code works for ymd96 as well. i have modified your code in my personal repo. however i am facing above mentioned issue in my board. hope you can look into and help.
1 thing i note that num lock key enables wrong led, it enables the caps led and not the assigned numlock key. hope this would give you some clue as to what may be wrong.

krusli, 2018-03-26:
For changing which LEDs turn on and off on caps lock/num lock, you could just swap the #define-s in [backlight.c](https://github.com/qmk/qmk_firmware/blob/master/keyboards/jj40/backlight.c).

Originally, it was

```C
#define NUMLOCK_PORT    (1 << 1)  // 1st pin of Port D (digital)
#define CAPSLOCK_PORT   (1 << 2)  // 2nd pin
```

Change that to

```C
#define NUMLOCK_PORT    (1 << 2)  // 2nd pin of Port D (digital)
#define CAPSLOCK_PORT   (1 << 1)  // 1st pin
```

As for the caps lock key getting stuck, was that problem already there before integrating the JJ40 backlighting code? I’ll try to check the keyboards’ directories and see if I could track down the problem once I have some time.

drashna, 2018-03-27:
`let_set_*` gets called every time a "lock" status changes (caps, num, scroll, etc).  Check that and see what's happening there. 

This may be in the main keyboard file, or in the keymap.

harshitgoel96, 2018-03-27:
@drashna @krusli 

I tried a few things and here is the result

Setting

`#define CAPSLOCK_PORT   (1 << 1)`
`#define NUMLOCK_PORT    (1 << 2)`

Make the caps lock work as expected, but now Num lock does not work. I tried some other pins 

1<<3 results in same error, 1<<5 no numlock led but toggle is working fine

I am guess either led_set_user results in some error or the pin 2 needs something more for lighting it up.  I will drop an email to vincent to know what is the correct pin for numlock led.

harshitgoel96, 2018-03-28:
@krusli  I have seen the code where you mention jj40 sets the NUM LOCK using i2c. 

I was able to get the schema from vincent for the ymd96. Looks like NUM lock and Caps lock are both attached directly to pins. Could this be issue for the code ?

[image]

Or should I just change numlock pin position as follows ?

`#define NUMLOCK_PORT (1 << 0)`

krusli, 2018-03-28:
That should be it - the schematic seems to indicate that the numlock LED is set in PD0. From what I remember I2C is only used for turning on and changing the colours of the underglow LEDs.

harshitgoel96, 2018-03-28:
@krusli so i just use `#define NUMLOCK_PORT (1 << 0)` ?

closed 2018-03-28 by harshitgoel96

StitiS, 2018-06-08:
Does the backlight still work for you?  I just cloned from "harshitgoel96/qmk_firmware" and the backlight isn't turning on when I hit "Raise + F2," or "Raise + Space" for example.  I know my backlight LEDs work because when I swap the backlight with the NumLock LED in backlight.c, they all turn on normally.

harshitgoel96, 2018-06-08:
it still works yes. can you double check the keymap you used once?

i will rebuild and try it again.

harshitgoel96, 2018-06-08:
@StitiS can you please confirm it is YMD96(2017) and NOT Melody96 (2018)  as both are different PCBs

harshitgoel96, 2018-06-08:
@StitiS  also please make sure you are using correct variant to build the keyboard, the fn+f2 and fn+space are defined on hgoel89 folder. 

`make ymd96:hgoel89`

harshitgoel96, 2018-06-08:
@StitiS  please take the latest commit from the  "harshitgoel96/qmk_firmware"  branch, it should be working fine now. please use hgoel89 variant to build the firmware

StitiS, 2018-06-08:
@harshitgoel96 Awesome thanks a bunch, that worked perfectly!

My YMD96 works great now.  Do you know what changed?  Was it a change that you had made locally, or something that was made in the main qmk_firmware repo?

harshitgoel96, 2018-06-08:
Something got messed up when I merged the local with git Branch. I have merged again from local, tested it and pushed it to hit.

StitiS, 2018-06-08:
Great thanks for the fix. As mentioned, it feels good to solder together
the keyboard and have it all work great.