qmk/qmk_firmware #2537: JC65

deadencryption, 2018-03-15:
CTRL and CAPS Lock reversed from code.
LGUI also no longer working.

Code below was working fine before:

#include "v32u4.h"

// How long (in ms) to wait between animation steps for the breathing mode
const uint8_t RGBLED_BREATHING_INTERVALS[] PROGMEM = {30, 20, 10, 5};

// How long (in ms) to wait between animation steps for the rainbow mode
const uint8_t RGBLED_RAINBOW_MOOD_INTERVALS[] PROGMEM = {120, 60, 30};

// How long (in ms) to wait between animation steps for the swirl mode
const uint8_t RGBLED_RAINBOW_SWIRL_INTERVALS[] PROGMEM = {30, 20, 3};

// How long (in ms) to wait between animation steps for the snake mode
const uint8_t RGBLED_SNAKE_INTERVALS[] PROGMEM = {100, 50, 20};

// How long (in ms) to wait between animation steps for the knight modes
const uint8_t RGBLED_KNIGHT_INTERVALS[] PROGMEM = {127, 63, 31};

// These control which colors are selected for the gradient mode
const uint16_t RGBLED_GRADIENT_RANGES[] PROGMEM = {360, 240, 180, 120, 90};

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
  [0] = KEYMAP(
      KC_GESC, KC_1, KC_2, KC_3, KC_4, KC_5, KC_6, KC_7, KC_8, KC_9, KC_0, KC_MINS, KC_EQL, KC_BSLS, KC_GRV, KC_PSCR,
      KC_TAB, KC_Q, KC_W, KC_E, KC_R, KC_T, KC_Y, KC_U, KC_I, KC_O, KC_P, KC_LBRC, KC_RBRC, KC_BSPC, KC_DEL,
      KC_CAPS, KC_A, KC_S, KC_D, KC_F, KC_G, KC_H, KC_J, KC_K, KC_L, KC_SCLN, KC_QUOT, KC_ENT, KC_PGUP,
      KC_LSFT, KC_NUBS, KC_Z, KC_X, KC_C, KC_V, KC_B, KC_N, KC_M, KC_COMM, KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_PGDN,
      KC_LCTL, KC_LGUI, KC_LALT, KC_SPACE, KC_SPACE, KC_SPACE, KC_RALT, MO(1), KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT
      ),
  [1] = KEYMAP(
      RESET, KC_F1, KC_F2, KC_F3, KC_F4, KC_F5, KC_F6, KC_F7, KC_F8, KC_F9, KC_F10, KC_F11, KC_F12, KC_INS, KC_PAUS, KC_TRNS,
      KC_TRNS, RGB_TOG, RGB_M_P, RGB_M_B, RGB_M_R, RGB_M_SW, RGB_M_SN, RGB_M_K, RGB_M_X, RGB_M_G, RGB_MOD, KC_VOLU, KC_VOLD, KC_MUTE, KC_TRNS,
      KC_TRNS, RGB_HUI, RGB_SAI, RGB_VAI, RGB_HUD, RGB_SAD, RGB_VAD, KC_DOWN, KC_HOME, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS,
      KC_TRNS, BL_TOGG, BL_STEP, BL_INC, BL_DEC, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_END, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS,
      KC_TRNS, KC_TRNS, KC_MPRV, KC_MPLY, KC_MPLY, KC_MPLY, KC_MNXT, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS
      ),
};

deadencryption, 2018-03-15:
Also noticed this now when compiling. I don't know if it's part of the issue:

$ make jc65:dead_encryption:dfu
QMK Firmware 0.5.248
Making jc65/v32u4 with keymap dead_encryption and target dfu

avr-gcc.exe (AVR_8_bit_GNU_Toolchain_3.5.4_1709) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   21898       0   21898    558a ./.build/jc65_v32u4_dead_encryption.hex

Copying jc65_v32u4_dead_encryption.hex to qmk_firmware folder                                       [OK]
Checking file size of jc65_v32u4_dead_encryption.hex                                                [OK]
 * File size is fine - 21898/28672
Bootloader Version: 0x00 (0)
Erasing flash...  Success
Checking memory from 0x0 to 0x6FFF...  Empty.
Checking memory from 0x0 to 0x55FF...  Empty.
0%                            100%  Programming 0x5600 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x7000 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
Validating...  Success
0x5600 bytes written into 0x7000 bytes memory (76.79%).
make[1]: *** [tmk_core/avr.mk:143: dfu] Error 127
Make finished with errors
make: *** [Makefile:543: jc65:dead_encryption:dfu] Error 1

drashna, 2018-03-15:
Errors look fine (eg, not actually critical).

For the Ctrl - CAPS, bootmagic strikes again. 

https://docs.qmk.fm/feature_bootmagic.html#bootmagic-keycodes

If bootmagic is enabled for the board, hold space and capslock when plugging it in. 

For the LGUI, does it work as a different key?  If not, hold space and LGUI when plugging it in. Otherwise, hold space and LCTRL when plugging it in.

deadencryption, 2018-03-15:
Holding down those keys does nothing for me. The board just loads normally.

deadencryption, 2018-03-15:
Ok, so I had to enable bootmagic on the rules.mk under jc65\v32u4\rules.mk 

Afterwards, your recommendations to hold win key plus spc worked and holding ctrl plus spc worked...any idea how the board ended up that way? 

Should I disable it again to prevent it from changing?

drashna, 2018-03-15:
Likely, the eeprom had some random values in it, that caused this. Resetting the EEPROM would probably have fixed this as well.

drashna, 2018-03-25:
Closing since this is resolved. 

And yes, I would disable bootmagic, if you don't need it. 

In fact, it may be better to add an eeprom reset key....

closed 2018-03-25 by drashna