qmk/qmk_firmware #2549: Some sort of error when erasing memory on Windows, flashing preonic rev2

Boy-314, 2018-03-17:
```
$ dfu-programmer atmega32u4 erase --force --debug 1000
     target: atmega32u4
    chip_id: 0x2ff4
  vendor_id: 0x03eb
    command: erase
      quiet: false
      debug: 1000
device_type: AVR
------ command specific below ------
   validate: true

dfu.c:414: dfu_device_init( 1003, 12276, 0063FE88, true, false )
dfu.c:431: found device at USB:1,0
dfu.c:663: Found DFU Inteface: 0
dfu.c:293: dfu_abort( 0063FE88 )
dfu.c:204: dfu_get_status( 0063FE88, 0063FE34 )
dfu.c:230: ==============================
dfu.c:232: status->bStatus: OK (0x00)
dfu.c:233: status->bwPollTimeout: 0x0001 ms
dfu.c:235: status->bState: dfuIDLE (0x02)
dfu.c:236: status->iString: 0x00
dfu.c:237: ------------------------------
dfu.c:688: State: dfuIDLE (2)
commands.c:87: erase 0x6FFF bytes.
atmel.c:444: atmel_erase_flash( 0063FE88, 4 )
Erasing flash...  dfu.c:142: dfu_download( 0063FE88, 3, 0063FDB1 )
dfu.c:163: Message: m[0] = 0x04
dfu.c:163: Message: m[1] = 0x00
dfu.c:163: Message: m[2] = 0xff
dfu.c:204: dfu_get_status( 0063FE88, 0063FDB4 )
dfu.c:230: ==============================
dfu.c:232: status->bStatus: OK (0x00)
dfu.c:233: status->bwPollTimeout: 0x0001 ms
dfu.c:235: status->bState: dfuIDLE (0x02)
dfu.c:236: status->iString: 0x00
dfu.c:237: ------------------------------
Success
atmel.c:494: CMD_ERASE status: Erase Done.
atmel.c:934: atmel_blank_check( 0063FE88, 0x00000000, 0x00006FFF )
atmel.c:1041: atmel_select_memory_unit( 0063FE88, 0 )
atmel.c:1055: Ignore Select Memory Unit for non GRP_AVR32 device.
Checking memory from 0x0 to 0x6FFF...
atmel.c:1100: atmel_select_page( 0063FE88, 0 )
atmel.c:1114: Selecting page 0, address 0x0.
dfu.c:142: dfu_download( 0063FE88, 4, 0063FD44 )
dfu.c:163: Message: m[0] = 0x06
dfu.c:163: Message: m[1] = 0x03
dfu.c:163: Message: m[2] = 0x00
dfu.c:163: Message: m[3] = 0x00
dfu.c:204: dfu_get_status( 0063FE88, 0063FD44 )
dfu.c:230: ==============================
dfu.c:232: status->bStatus: OK (0x00)
dfu.c:233: status->bwPollTimeout: 0x0001 ms
dfu.c:235: status->bState: dfuIDLE (0x02)
dfu.c:236: status->iString: 0x00
dfu.c:237: ------------------------------
atmel.c:867: __atmel_blank_page_check( 0063FE88, 0x00000000, 0x00006fff )
dfu.c:142: dfu_download( 0063FE88, 6, 0063FDAE )
dfu.c:163: Message: m[0] = 0x03
dfu.c:163: Message: m[1] = 0x01
dfu.c:163: Message: m[2] = 0x00
dfu.c:163: Message: m[3] = 0x00
dfu.c:163: Message: m[4] = 0x6f
dfu.c:163: Message: m[5] = 0xff
dfu.c:204: dfu_get_status( 0063FE88, 0063FDB4 )
dfu.c:230: ==============================
dfu.c:232: status->bStatus: errCHECK_ERASED (0x05)
dfu.c:233: status->bwPollTimeout: 0x0001 ms
dfu.c:235: status->bState: dfuIDLE (0x02)
dfu.c:236: status->iString: 0x00
dfu.c:237: ------------------------------
atmel.c:902: Region is NOT bank.
dfu.c:180: dfu_upload( 0063FE88, 2, 0063FDAC )
atmel.c:910:  First non-blank address in region is 0x0.
atmel.c:981: Flash NOT blank beginning at 0x0.
Not blank at 0x1.
```

drashna, 2018-03-25:
MSYS, WSL, Docker? 

And continuing to happen?

Boy-314, 2018-03-25:
MSYS. continuing to happen. From what I can tell the MCU needs to be ISP flashed a new bootloader

drashna, 2018-03-27:
That's entirely possible. 

If you do ISP flash it... run `make preonic/rev2:boy314:production` and flash the hex that this produces.  That contains both the bootloader (QMK DFU) ***and*** the firmware hex.  So you shouldn't need to do anything else.

Boy-314, 2018-03-27:
If I had the proper equipment to do it myself I would, but as a student I'm quite limited. Our labs use's some pretty janky solder. Thanks for confirming the issue.

drashna, 2018-03-27:
If you have a spare controller, then you have an ISP flasher:
https://docs.qmk.fm/isp_flashing_guide.html

It looks like the pins are available, so you could ISP flash it, if you can hook it up to the holes on the board (GND, PB1-3, VCC, RST (next to the actual button))

Not sure if that helps, but ....

closed 2018-10-22 by drashna