qmk/qmk_firmware #2580: "Bug" in avr-gcc causing internal compiler error.

cschwenz, 2018-03-21:
This needs to be documented somewhere because internet searches for this issue returned nothing for me.

Please note, `avr-gcc` is doing the correct thing. Updating/upgrading tangentially related software packages from `brew` can cause sym-links `avr-gcc` depends on to be mangled.

So, if you see something like:
```
% make viterbi:default:avrdude
Making viterbi/rev1 with keymap default and target avrdude

avr-gcc (GCC) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  20696	      0	  20696	   50d8	./.build/viterbi_rev1_default.hex

Compiling: keyboards/viterbi/matrix.c
                                                         dyld: Library not loaded: /usr/local/opt/isl/lib/libisl.15.dylib
  Referenced from: /usr/local/Cellar/avr-gcc/7.3.0/libexec/gcc/avr/7.3.0/cc1
  Reason: image not found
avr-gcc: internal compiler error: Abort trap: 6 (program cc1)
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
 [ERRORS]
 |
 |
 |
make[1]: *** [.build/obj_viterbi_rev1_default/matrix.o] Error 1
make: *** [viterbi:default:avrdude] Error 1
Make finished with errors
```

The solution is to do:
```
brew rm avr-gcc
brew rm dfu-programmer
brew rm gcc-arm-none-eabi
brew rm avrdude
brew install avr-gcc
brew install dfu-programmer
brew install gcc-arm-none-eabi
brew install avrdude
```

Ideally, this needs to be added to either the https://docs.qmk.fm/getting_started_build_tools.html page or the https://docs.qmk.fm/faq_build.html page.

drashna, 2018-03-26:
Added this in #2611, to the build/compile FAQ, as many errors are listed there.

Thanks!

closed 2018-03-26 by drashna

earboxer, 2019-08-03:
I recently got this error (`dyld: Library not loaded: /usr/local/opt/isl/lib/libisl.19.dylib`
), and had to `brew reinstall avr-gcc@7` for it to work.

fauxpark, 2019-08-03:
@earboxer - you can actually upgrade to avr-gcc@8 now.