qmk/qmk_firmware #2601: [request] bootloader option for atmega32a

harshitgoel96, 2018-03-24:
I know there are different bootloader that can be used. For a noob like me its a very deep sea. I can see there is already bootloader shipped for atmega32u4 MCUs. Can there be option so support more MCU? or if we can have guide for boot loaders.

drashna, 2018-03-24:
Well, that's a complicated issue.   For the most part, the firmware itself doesn't care.  There are some additionally handling based on the bootloader, in some cases .... but .....

DFU is probably the most common, and probably the simplest to flash from script.   Many use Caterina (Arduino, especially the Pro Micro). PJRC boards (Teesny's) use their own bootloader (halfkay). BootloadHID is a nead one. 

From there, there are a few others. 
And which the keyboard board uses can be configured in it's rules.mk file. 

That said, to switch the bootloader, you have to "ISP flash" the board. Which isn't exactly simple, and can be easily screwed up (it shouldn't perma-brick the board, but you may have a time of it figuring out what went wrong)

And there is a very basic guide on how to ISP flash your MCU: 
https://docs.qmk.fm/isp_flashing_guide.html

harshitgoel96, 2018-03-24:
Thanks for the fast response. I have a ymd96 on which I managed to kill the Bootloader. I have been able to flash the qmk hex using the ISP programer. Now I need to rebuild my Bootloader, and there is no good guide on it. Hence the request about it.

drashna, 2018-03-24:
Well, most of the bootloaders are a bit outside of what QMK does.  Which means that the docs are a bit "anemic"

It looks like the atmega32a is basically the same as the atmega32u4, but without native USB support.  

There is the QMK DFU bootloader, and that *may* work, but I'm not sure.  @jackhumbert would have to confirm that it works for the atmega32a or not (the controller in the ymd96).
Or you could test it out (if you add `:production`, the hex generated should contain the bootloader and firmware.  If that doesn't work, then go back to just the firmware.

Also, there should be no issue with just ISP flashing the firmware, other than it being a PITA to flash.

drashna, 2018-03-24:
---

And maybe it would be a good idea to host the bootloaders for other chips, just in case. 
That or expand the docs for the QMK DFU bootloader, and make sure it supports chips like the ATMega32a, which need virtual USB support (or confirm that it does)

harshitgoel96, 2018-03-29:
@drashna after searching for quite a while, i have finally found the bootloaderHID for ps2avrgb.  Should I add it to the repo ? and if yes, where should i put it up ?

drashna, 2018-03-29:
For now, if you could link it here. 
As for adding it to the repo.... that may be an IP/copyright issue.  A link in the docs would be better.  

Maybe a section in the ISP flashing guide for all of the bootloaders, even.

harshitgoel96, 2018-03-30:
Adding ps2avrgb bootloader

http://blog.winkeyless.kr/m/152

closed 2019-12-21 by stale[bot]