qmk/qmk_firmware #2610: Sending the same keycode multiple times, fast, with different physical keys.

dragonfax, 2018-03-25:
This is a request for help on configuration.

I have multiple keys send the same keycode (KC_BSPC). And I like to hit them all really fast in succession. Sending that same keycode multiple times.  I know I can have one key auto-repeat, but thats not what I'm after. I want to do it manually.

This works about 80% today. But when I'm hitting these keys fast, it leaves out a few of the keycodes. For example, if I hit 2 keys rapidly in success (key1 down, key2 down, key1 up, key2 up). It only sends a down/up for one keycode, i.e only one KC_BSPC is sent to the computer, not 2 of them.

Is this just debounce activating? Or is this some other detail in the design. Is the debounce done on the key in the matrix, or done on the keycode your sending in the end? 

What else could I do to fix this loss of keycodes sent?

skullydazed, 2018-03-25:
So here's the process for turning key presses into usb reports:

* Your keyboard matrix is scanned
* The physical keys being held down are translated into an array of held keys, based on the key code name
* This array is then compared against the last such array and key up and key down events are generated based on this comparison

Unpacking that, here's what happens in your scenario:

* Initial array: []
* key1 pressed, new array is generated: []
* Array is compared against the initial array, and comes up with a keydown for `KC_BSPC`
* key2 pressed, new array is generated: [`KC_BSPC`]
* No changes, so no usb report
* key1 released, new array is generated: [`KC_BSPC`] (because key2 is still being held down)
* No changes, so no usb report
* key2 released, new array is generated: []
* Array is compared against the last array and a keyup for `KC_BSPC` is sent

I think to have it work the way you want you'll need to replace `KC_BSPC` with a macro that sends backspace. You can read about defining macros here: https://docs.qmk.fm/feature_macros.html

dragonfax, 2018-03-25:
I'll give that a try and report back.

drashna, 2018-03-25:
That or use tapdance to set how many times it sends it.

dragonfax, 2018-03-26:
Seems to do the trick.
https://github.com/dragonfax/qmk_firmware/commit/55cdae3440ae9dfe49587115b95477dcf1b7f127

Thanks.

closed 2018-03-26 by dragonfax

drashna, 2018-03-26:
Awesome, glad to hear it!

fredizzimo, 2018-04-07:
I think that this https://github.com/qmk/qmk_firmware/pull/2710, should also fix this issue.