qmk/qmk_firmware #2616: Nesting advanced keycode?

ArchangeGabriel, 2018-03-26:
Is it possible to nest (some) advanced keycodes?

For instance, I’d like to do the following: `LT(FN, OSM(MOD_LCTL))`.

drashna, 2018-03-26:
Unfortunately no.

It's something that I'd like to see, but no idea how to implement/fix, personally.

ArchangeGabriel, 2018-03-26:
OK, so they are currently no way to mix a layer and a modifier key together? Actually I realized that what I wanted was the opposite: `MT(MOD_LCTL, OSL(FN))`, but I guess it’s no as well.

drashna, 2018-03-26:
Not without writing your own keycodes, at least.  Which is possible, actually.  The docs include some code on how to make that possible: 
https://docs.qmk.fm/feature_advanced_keycodes.html#shifted-keycodes
And: 
https://docs.qmk.fm/feature_macros.html

However, this would be on the complicated side.  You'd probably have to set up timers, and use the matrix scan code, etc.  Not sure if you feel that it's worth the trouble.

ArchangeGabriel, 2018-04-01:
No indeed, I will rather change my layout to avoid requiring this.