qmk/qmk_firmware #2618: :production results in compilation error for bootloaderHID based keyboards

harshitgoel96, 2018-03-26:
I am working on YMD96 and getting following error while making the keyboard with :production param.
this is also true when making jj40.
```
$ make ymd96:krusli:production //this is custom keyboard mapping i am working on. Same result when using jj40:krusli:production and ymd96:default:production.
QMK Firmware 0.5.264
WARNING:
 Some git sub-modules are out of date or modified, please consider runnning:
 make git-submodule
 You can ignore this warning if you are not compiling any ChibiOS keyboards,
 or if you have modified the ChibiOS libraries yourself. 

Making ymd96 with keymap krusli and target production

avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  22198	      0	  22198	   56b6	./.build/ymd96_krusli.hex

Compiling: ./tmk_core/common/command.c                                                              [OK]
Linking: .build/ymd96_krusli.elf                                                                    [OK]
Creating load file for flashing: .build/ymd96_krusli.hex                                            [OK]
 [RM]      : Removing object files of "BootloaderDFU"
 [RM]      : Removing dependency files of "BootloaderDFU"
 [RM]      : Removing output files of "BootloaderDFU"
 [INFO]    : Begin compilation of project "BootloaderDFU"...

avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

 [GCC]     : Compiling C file "BootloaderDFU.c"
In file included from ../../LUFA/../LUFA/Drivers/Board/AVR8/QMK/LEDs.h:106:0,
                 from ../../LUFA/../LUFA/Drivers/Board/LEDs.h:138,
                 from BootloaderDFU.h:55,
                 from BootloaderDFU.c:37:
./Keyboard.h:1:1: error: expected identifier or ‘(’ before ‘-’ token
 -e #ifndef QMK_KEYBOARD
 ^
./Keyboard.h:1:4: error: stray ‘#’ in program
 -e #ifndef QMK_KEYBOARD
    ^
./Keyboard.h:4:4: error: stray ‘#’ in program
 -e #define MANUFACTURER ymdk china
    ^
./Keyboard.h:5:4: error: stray ‘#’ in program
 -e #define PRODUCT ymd96 Bootloader
    ^
./Keyboard.h:11:2: error: #endif without #if
 #endif
  ^
BootloaderDFU.c: In function ‘SetupHardware’:
BootloaderDFU.c:244:2: warning: implicit declaration of function ‘LEDs_Init’ [-Wimplicit-function-declaration]
  LEDs_Init();
  ^
../../LUFA/Build/DMBS/DMBS/gcc.mk:216: recipe for target 'obj/BootloaderDFU.o' failed
make[2]: *** [obj/BootloaderDFU.o] Error 1
tmk_core/avr.mk:253: recipe for target 'bootloader' failed
make[1]: *** [bootloader] Error 2
Makefile:534: recipe for target 'ymd96:krusli:production' failed
make: *** [ymd96:krusli:production] Error 1
Make finished with errors

```

drashna, 2018-03-26:
`:production` creates a fully "ISP" ready hex, that contains both the QMK firmware and QMK DFU bootloader. IIRC. 

If you don't have the QMK_ options defined, it will error out, as these are necessary.

drashna, 2018-03-26:
Docs about it here: https://docs.qmk.fm/flashing.html

And you can see what it's looking for when compiling:
https://github.com/qmk/qmk_firmware/blob/master/tmk_core/avr.mk#L255-L260

You can see this in the Planck
https://github.com/qmk/qmk_firmware/blob/master/keyboards/planck/config.h#L39-L42

Also, it will still error out, as @jackhumbert may have missed something in the build script.  A normal hex isn't made, but it is looked for in the size check:
https://github.com/qmk/qmk_firmware/blob/master/tmk_core/avr.mk#L270
(i'd change this, but I'm not sure how ... other than to remove the first target

harshitgoel96, 2018-03-29:
I tried setting up the QMK_* options, still getting same error. I think this is due to custom led driver.

drashna, 2018-03-29:
Yeah, it probably is, because I'm not seeing the issue on my viterbi.

harshitgoel96, 2018-03-30:
I think since bootloaderhid is not supported for :production param, can someone add check for it instead ?

drashna, 2018-05-11:
Tested this out, works for me. Check your code, you probably have an extra character somewhere, or a missing char.

jackhumbert, 2020-03-17:
This looks like it was resolved, but I think it's something we need to document for clarification.

fauxpark, 2020-03-17:
For future reference: `:bootloader` and `:production` will build the QMK-DFU bootloader, but this will only work for MCUs supported by LUFA, as that's where the code is from: https://github.com/qmk/lufa/tree/master/Bootloaders/DFU
So at present there is no way to compile QMK-DFU for the ATmega32A, 328P, or any ARM chip. The first two have no native USB support (thus they use V-USB which does it all in software), and ARM is a completely different architecture. Most of the STM32 chips have a DFU bootloader in ROM anyway.

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc