qmk/qmk_firmware #2657: New project with Teensy 3.2 support?

huytd, 2018-04-02:
Hi, I'm having an issue with Teensy 3.2

I'm building a custom layout.

When I created a new project using `./utils/new_project.sh`, and flashing the compiled firmware, the computer did not recognize the keyboard.

Is Teensy 3.2 fully supported in QMK yet?

Thanks,
Huy

drashna, 2018-04-02:
It is.  But I think that the default file is geared more for AVR. 

You need to make sure that all of the settings are correct. 

I'm not an expert here, so somebody like @skullydazed may be better to help.  

But ... you may want something like: 
https://github.com/qmk/qmk_firmware/blob/ab197af2ead7c764aa48ee4a5aa7b1b9633ce5b7/keyboards/k_type/rules.mk

huytd, 2018-04-02:
@drashna Thanks you so much, I'll try this again!

huytd, 2018-04-02:
Great! with a bit modification, I'm able to get my board working now!

Thank you so much @drashna

closed 2018-04-02 by huytd

drashna, 2018-04-02:
Glad to hear it, and you're welcome!

DmitryMyadzelets, 2019-01-17:
> ... with a bit modification, I'm able ...
@huytd could you tell more what your modifications are, please? I have Teensy 3.2 and have no idea where to start from.

huytd, 2019-01-17:
@DmitryMyadzelets sorry man, it's quite a long time ago, I still ran into some trouble after that. I ended up writing my own firmware instead of using QMK.

DmitryMyadzelets, 2019-01-17:
@huytd, would you mind sharing? Can't find in your repos.

I'm new to keyboards, but feel ok with embedded stuff. Don't really want to start from ground though.

huytd, 2019-01-17:
It's this https://gist.github.com/huytd/8dabf762a868b86d2aa597b878e53df0

You gonna need Teensyduino to compile, don't forget to set the USB Type to Keyboard as commented in the file header.

This version is a 2x2 matrix, but basically you can expand it by changing the constants without the need to modify any other part of the code.

- Change the rows/cols here https://gist.github.com/huytd/8dabf762a868b86d2aa597b878e53df0#file-keyboard-prototype-c-L16-L17
- And change the key mapping matrix here https://gist.github.com/huytd/8dabf762a868b86d2aa597b878e53df0#file-keyboard-prototype-c-L24-L30

The two REGISTER_DELAY and REPEAT_DELAY are for debouncing https://gist.github.com/huytd/8dabf762a868b86d2aa597b878e53df0#file-keyboard-prototype-c-L18-L19

I'm using Cherry MX Brown, if you also using cherry switches, you probably don't have to modify these.

Hope it can help :)

huytd, 2019-01-17:
Oh, and it's lack of per-key-debouncing, modifiers and layout switching at this moment. But I think it's easy to implement these once we have a barebone code above.

DmitryMyadzelets, 2019-01-18:
Oh, great! Thank you, @huytd. I have installed the toolchain already and uploaded a first test to the Teensy. So far so good.

huytd, 2019-02-06:
@DmitryMyadzelets hey, if you're (and anyone) still interested, I'm just finished my keyboard, you can take a look at the code here https://github.com/huytd/snackymini-keyboard/blob/master/firmware/firmware.ino

It has 2 layouts and I think it's pretty easy to customize now.