qmk/qmk_firmware #2662: [help] Make Caps Lock LED turn off when I turn off the other backlight LEDS

FSund, 2018-04-03:
I have a DZ60-based board, and I sometimes leave the backlight off. In this case it would be nice to just disable the Caps Lock LED (disregarding the Caps Lock state). 
How can I program this?

Optionally, I would also like the Caps Lock LED to always be on when my backlight is on.

I think I could do this by physically connecting the Caps Lock LED to a neighboring LED on the PCB, but I'd rather do it programatically if possible.

drashna, 2018-04-05:
I think you'd want to use something like this: 

```c
void matrix_scan_user (void) {
  if (get_backlight_level) {
      PORTB &= ~(1 << 2);
  } else {
      PORTB |= (1 << 2);
  }
}
```

FSund, 2018-04-06:
Thanks! You probably meant `get_backlight_level()` and not `get_backlight_level`, but other than that it seems to work. 

One issue though, it only works if I decrease the backlight to zero using `BL_DEC`, but not if I toggle it using `BL_TOGG`. Could this be a bug in QMK?

Here is the code for toggling backlight (from `backlight.c`)

    void backlight_toggle(void)
    {
        backlight_config.enable ^= 1;
        if (backlight_config.raw == 1) // enabled but level = 0
            backlight_config.level = 1;
        eeconfig_update_backlight(backlight_config.raw);
        dprintf("backlight toggle: %u\n", backlight_config.enable);
        backlight_set(backlight_config.enable ? backlight_config.level : 0);
    }

FSund, 2018-04-06:
I'll see if I can get debugging/printing to work, and see if I can confirm that `backlight_config.level` is not set to zero by `BL_TOGG`, as I suspect. If that's the case I'll report this as a bug, as I'm unsure how to fix it.

See my comment above for more info.

drashna, 2018-04-06:
Well, glad I pointed you in the right direction, at least. 

As for fixing, check the `get_backlight_level()` thing, but you'd also want to check `backlight_config.enable` too. 

I think you want to add `extern backlight_config_t backlight_config` to your keymap.  I'm not 100% sure about this (maybe even just `backlight_config_t backlight_config`, and ... possibly `#include "backlight.h"`) 

Then use:  `if (!backlight_config.level || !backlight_config.enable)` and swap the PORTB things.  So it would be like: 
```c
void led_set_user(uint8_t usb_led) {
  if (!backlight_config.level || !backlight_config.enable) {
    PORTB |= (1 << 2);
  } else {
    PORTB &= ~(1 << 2);
  }
}```

So it disables the light if the level is 0, OR if it's not enabled.  That should fix the behavior.

drashna, 2018-04-11:
Awesome, glad to hear it!