qmk/qmk_firmware #2664: Caps Lock key responding slowly

NateEag, 2018-04-03:
I recently tried binding KC_CAPS to a key, having discovered to my surprise that I do actually want Caps Lock occasionally.

It is not responding as quickly as regular keys in my keymap, and I don't know why.

If I tap it at my usual typing speed, no keyboard event gets sent for the keypress (based on Karabiner Elements' EventViewer.app).

If I hold it just slightly longer than I normally would, then the event is sent and Caps Lock is activated as I would expect.

Interestingly, I experience no lag turning it off. The key seems just as responsive doing that as any of the other keys.

I'm suspicious that perhaps my keymap is causing this somehow but I haven't had the time to try a reproduction with a simple keymap yet. I eyeballed the whole thing and nothing jumped out at me, but I'm no expert in QMK or C.

My keymap can be found [here](https://github.com/NateEag/qmk_firmware/tree/add-truly-ergodoxic-layout). It's a few months behind master.

drashna, 2018-04-03:
That's really odd.  I tested it out on my board (Ergodox EZ) and it seems to work just fine.  Registers immediately (both for the LEDs on my board, and on the system, Windows). 

It may be worth testing using one of these methods: https://docs.qmk.fm/newbs_testing_debugging.html#testing

Though looking at things, it .... may be a side effect of the higher debouncing setting?  
Otherwise I can't see anything.

So it may be worth merging/rebasing your repo, so it's more up to date.

skullydazed, 2018-04-03:
This is a feature in macOS to prevent accidental presses. You have to hold the key for at least 200ms or the OS ignores the input.

drashna, 2018-04-03:
That's good to know :)

NateEag, 2018-04-04:
Wow.

Thanks, @skullydazed - it would have taken me a long, long time to tumble to this.

Now to figure out how I can deactivate that 'feature'...

NateEag, 2018-04-04:
I found [this rather involved solution](https://apple.stackexchange.com/a/199958) on Stack Exchange, but I didn't find anything related to the ErgoDox in step 1, so I'm not sure how to progress right now.

Hopefully I'll find time to look into it soon.

drashna, 2018-05-11:
Mind if I close this, since it appears resolved

NateEag, 2018-05-15:
Closing.

If I ever do find a reasonable way to remove the delay on OS X, I'll update the thread for completeness' sake, but you're right, this does not need to be an open issue for QMK.

closed 2018-05-15 by NateEag

minjaep, 2021-04-02:
Just found this while trying to find a fix. I actually figured out if you use mod-tap then it works without delay. That is, try something like MT(MOD_LGUI, KC_CAPSLOCK) instead of KC_CAPSLOCK.

NateEag, 2021-04-05:
Thanks, @minjaep! I have yet to find a solution to this.

I've made a note to give this a try.

pspEgg, 2021-10-08:
> Just found this while trying to find a fix. I actually figured out if you use mod-tap then it works without delay. That is, try something like MT(MOD_LGUI, KC_CAPSLOCK) instead of KC_CAPSLOCK.

I tried this in VIA, there is improvement but not enough for me, so I tried using a Macro and it has been most reliable.

how to:
1. define a macro with {KC_CAPS}
2. assign that macro to the capslock location

NateEag, 2022-03-08:
I'm currently using @pspEgg's workaround in my layout and it's working like a charm. Thanks!