qmk/qmk_firmware #2670: Add support for Input:Club serial numbers

belak, 2018-04-04:
As mentioned in https://github.com/qmk/qmk_firmware/pull/1963#issuecomment-342670875 it should be possible to expose the I:C serial numbers. Now that we've got a common IC_TEENSY board, maybe it would make this a little easier? I'm not 100% sure how that portion of the codebase works though. It might be harder than expected to set the USB serial.

yiancar, 2018-05-15:
So I can tell you how to set the serial number.
The define is in tmk_core/protocol/usb_descriptor.c

Setting in your config #define SERIAL_NUMBER should do it.

Now do you know where in eeprom the serial number is saved?

belak, 2018-05-15:
This shows how they pull the information out of the register in the official firmware:

https://github.com/kiibohd/controller/blob/1a79a7f8956fa94f232c60787a672015e55e94b3/Output/USB/arm/usb_dev.c#L1743

Here's where the register locations are defined:

https://github.com/kiibohd/controller/blob/0203b275e9bf16b5743d417e25295b366f9ff95a/Lib/kinetis.h#L456

yiancar, 2018-05-15:
I would suggest trying to pull it off the eeprom and display it with xprint as debug to make sure u cn actually pull it.

If that works we will go from there

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc