qmk/qmk_firmware #2683: Flashing DZ60 on Windows 7 requires a USB driver

gwerbin, 2018-04-05:
On Windows 7, DFU Programmer (via QMK Toolbox) doesn't recognize the DZ60 until the `libusbK` driver is installed.

I used a program called [Zadig](http://zadig.akeo.ie/) to install it, which was about as painless as could be. Just select `libusbK` from the drop-down and press the "Install" button.

tsmanikandan, 2018-04-05:
Same issue for the NIU mini. Zadig fix works out

mechmerlin, 2018-05-01:
When using QMK Toolbox on Windows, please install the recommended drivers first. You can install them all using the install all drivers batch script in

https://github.com/qmk/qmk_driver_installer/releases

drashna, 2018-10-22:
This was added to the docs: 
https://docs.qmk.fm/#/faq_build?id=unknown-device-for-dfu-bootloader

closed 2018-10-22 by drashna

gwerbin, 2019-02-06:
I recently fixed a problem with a Tokyo60 caused by installing the *wrong* drivers, and the fix might be instructive for others.

### Symptoms

- Keyboard worked during boot, but stopped sending characters once Windows loaded
- RGB controls (anything that didn't require sending data to the computer) worked fine
- The keyboard was recognized when reset to DFU mode. Flashing and EEPROM clearing appeared to work correctly
- Keyboard appeared in Device Manager under "libusbK devices" instead of under "Keyboards"

### Solution

1. Right-click the keyboard(s) in Device Manager, and click "Uninstall".
2. Check the box "Delete the driver software for this device".
3. Click OK to proceed.
4. Unplug the keyboard.
5. Plug it back in. It should work fine now.

I had previously wrestled with this issue on Discord, and the "delete the driver software" step is what I was missing.

EchoSunset, 2021-06-14:
This did not work for me. My DZ60 separated into 3 interfaces. One is detected as a keyboard, one as a libusbK, and one as a libusb-win32 device. Any recommendations?

fauxpark, 2021-06-14:
https://docs.qmk.fm/#/driver_installation_zadig?id=recovering-from-installation-to-wrong-device

EchoSunset, 2021-06-14:
> https://docs.qmk.fm/#/driver_installation_zadig?id=recovering-from-installation-to-wrong-device

Thank you so much! I read the article before but did not realize that I would need to repeat the process.