qmk/qmk_firmware #2684: two satan boards give different keycode outputs for same matrix

arianvp, 2018-04-05:
Hey,,

I have two `SATAN GH60` boards (https://github.com/qmk/qmk_firmware/tree/master/keyboards/satan)

But one is outputing random keycodes whilst the other is functioning fine.
First I checked if the PCB differed. They did, but after looking at the traces, the pinout and the matrix were the same. Just some capacitors were moved.

Because one of the boards is outputting garbage, I force-enabled all debug modes in `bootmagic.c` so that the boards always print out their matrix status and keycodes

Now the interesting part. The key _matrix_ output is identical bu the _keycodes_ are different.   I flashed the same firmware to both..   Any idea how it can be that the matrix output is identical, but the keycode output is not?

 Attached you find the debug of pressing the buttons on the top row and the 2nd row:

<details>
<summary>A</summary>

```
r/c 0123456789ABCDEF
00: 0000000000000000
01: 0000000000000000
02: 0000000000000000
03: 0000000000000000
04: 0000000000000000
keyboard_report: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

r/c 0123456789ABCDEF
00: 1000000000000000
01: 0000000000000000
02: 0000000000000000
03: 0000000000000000
04: 0000000000000000
keyboard_report: 00 00 29 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

r/c 0123456789ABCDEF
00: 0000000000000000
01: 0000000000000000
02: 0000000000000000
03: 0000000000000000
04: 0000000000000000
keyboard_report: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

r/c 0123456789ABCDEF
00: 0100000000000000
01: 0000000000000000
02: 0000000000000000
03: 0000000000000000
04: 0000000000000000
keyboard_report: 00 00 1E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

r/c 0123456789ABCDEF
00: 0000000000000000
01: 0000000000000000
02: 0000000000000000
03: 0000000000000000
04: 0000000000000000
keyboard_report: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

r/c 0123456789ABCDEF
00: 0010000000000000
01: 0000000000000000
02: 0000000000000000
03: 0000000000000000
04: 0000000000000000
keyboard_report: 00 00 1F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

r/c 0123456789ABCDEF
00: 0000000000000000
01: 0000000000000000
02: 0000000000000000
03: 0000000000000000
[... 850 lines cut ...]
00: 0000000100000000
01: 0000000000000000
02: 0000000000000000
03: 0000000000000000
04: 0000000000000000

r/c 0123456789ABCDEF
00: 0000000000000000
01: 0000000000000000
02: 0000000000000000
03: 0000000000000000
04: 0000000000000000
keyboard_report: 00 00 5F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
keyboard_report: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

r/c 0123456789ABCDEF
00: 0000000100000000
01: 0000000000000000
02: 0000000000000000
03: 0000000000000000
04: 0000000000000000

```

</details>

drashna, 2018-05-11:
Sorry for the delay,

Sounds like you may not be using the correct keymap:
https://github.com/qmk/qmk_firmware/blob/master/keyboards/satan/satan.h

Also, is it maybe this board: https://github.com/qmk/qmk_firmware/tree/master/keyboards/gh60

arianvp, 2018-05-16:
Hmm does that explain the fact that they output the same Matrix in debug mode though?

cluzz, 2018-06-03:
It's this board: 

[image]
bought from banggood.

The PCB differs from the one that @arianvp has but it does perform the matrix scan correctly -- we verified that with the debugging mode. It must somehow be the content of the usb packets. If the chip is capable of correctly sending which key in the matrix is pressed, then it must be able to send correct usb packets. Therefore neither the hardware, nor the usb driver could be the issue.

drashna, 2018-10-22:
Is the OS using the English US layout?  

Otherwise, QMK sends what it's told to send.

jackhumbert, 2020-03-17:
I think this can attributed to hardware differences between the boards (there are a lot of GH60-based PCBs) - if you have any specific fix you encountered, feel free to share it here!

closed 2020-03-17 by jackhumbert