qmk/qmk_firmware #2714: Using shift with a shifted keycode makes keys send gobbledygook

parkerault, 2018-04-08:
I'm afraid I don't know which version of QMK I'm using as it's on a prebuilt ergodox EZ, but I am on osx 10.12.6. As stated in the issue title, when I try to hold shift and send a shifted punctuation code (parens in this case) the keyboard starts sending what I assume are numeric keycodes, and I have to reboot the board to return it to normal. I discovered this when trying to see if holding shift would cause the shifted punctuation to return to the non-shifted character, which would be awesome, but alas... Is this some sort of easter egg? 😄

drashna, 2018-04-08:
Are you using the Ergodox EZ configurator?   

If so, try redownloading the hex file from the website and see if that works better. 
If it doesn't help, could you link the specific built? 

If you're using QMK to edit and compile it yourself, it should so the version immediately after the make command. Eg:
```
QMK Firmware 0.6.5
```

If it doesn't, then run `git log` on the system and look at the "master" entry, and the commit number in front of that entry.

parkerault, 2018-04-08:
Oh duh, I didn't consider that I can link to the configurator... here it is: https://configure.ergodox-ez.com/keyboard_layouts/kvodzr/edit

drashna, 2018-04-08:
I downloaded the hex and flashed it. I couldn't reproduce the issue that you have. It's .... really odd. 
But I'm using Windows, in case that matters. 

Also, try: https://drive.google.com/open?id=13yhJxzBSI1GGztOEUCqhVKZfFod_IpTu

rduenasf, 2018-04-18:
I've also experienced the same issue with ergodox-ez by using an old firmware from their configurator. 

I've been able to repro on OSX (not sure if it happens on windows) with:
* Flash [this]( https://www.dropbox.com/s/9psfx0hddsl1bow/ergodox_ez_firmware_qgmvgr_rd_20170307_v1.hex?dl=0) `.hex` file. This hex was generated on March 7th 2017.
* Having space-cadet shift for left-shift and right-shift
* Type `(` `)` very fast by using space-cadet shifts or at the same time. Real example: typing `foo()`.
* Type anything else and you'll see some crazy random character spam coming out until you reboot the keyboard (by unplugging in my case).

By playing around with the firmware and compiling it myself trying to isolate the issue I think this was solved in QMK by commit https://github.com/qmk/qmk_firmware/commit/13c882a1fbd3fe3d4f703628df46f18fc69e09ea

This prevents the keyboard from going into spam mode but has the side effect that while fast typing `(` and `)` they cancel each other.

seanniu93, 2018-05-02:
This issue also affected me on Linux using an ErgoDox EZ. For me it only occurs using **right** shift and **open** paren. The first open paren is not inputted. Subsequent keypresses either result in a numeric code, no input (possibly modifier key), or even a media key (like mute). I haven't exhaustively tested every key. The issue can be fixed by power cycling the board.

Right shift + close paren is also abnormal and results in the first close paren not being typed and repeated parens inputting correct. Typing letters while still holding down shift results in lowercase letters.

Left shift + either paren works as expected: the paren is inputted.

bendk, 2018-05-04:
I'm experiencing this one as well with:
  - linux with an ergodox EZ
  - Right shift only
  - Any of these chars: "!@#$%^&*" 

I'm building from commit 67053712f885641d3a8332239416ff8163a68647

bendk, 2018-05-04:
Here's the issue, from qmk_firmware/keyboards/ergodox_ez/config.h:

````
/* key combination for command */                                                                                                                                                                                  
#define IS_COMMAND() ( \                                                                                                                                                                                           
    keyboard_report->mods == (MOD_BIT(KC_LCTL) | MOD_BIT(KC_RCTL)) || \                                                                                                                                            
    keyboard_report->mods == (MOD_BIT(KC_LSFT) | MOD_BIT(KC_RSFT)) \                                                                                                                                               
)                                                                                                                                                                                                                  
```

- If both left and right shift are down, then it treats the next key as a command.
- This happens when you hold right shift, then hit a key with QMK_LSFT like KC_EXLM and friends
- Command+Number switches the default layer, so after this action you are now stuck in the symbol layer or something with no obvious way to get back

I think there are two solutions:
  - Make it so only left ctrl+right ctrl starts command mode
  - Make it so if right shift is down, KC_EXLM and friends don't register left shift

The second solution seems pretty nice to me, but some users might depend on left-shift being hit.  For now I'm just going to change IS_COMMAND in my config.h file.  If someone who knows more thinks that solution is good, I can try to make a patch.

bendk, 2018-05-07:
Maybe the best way would be to change the configurator to set COMMAND=no in the rules.mk file.  It seems like the command stuff is pretty arcane and I personally had no clue about it.

DanielGGordon, 2018-05-11:
@bendk but that doesn't explain why this issues is only occurring in non-Windows environments and with an ergodox-ez only.

bendk, 2018-05-11:
Different keyboards define IS_COMMAND differently.  I'm guessing those not affected have a different definition.  I'm not sure about the windows issue, but I think it might just be a coincidence.  Has anyone tested the same keyboard with different OSes?

drashna, 2018-10-22:
@DanielGGordon It may be how the OS was handling the input. 
@bendk And not all boards have COMMAND enabled, as well.

bendk, 2018-10-22:
Can I ask why this one was closed?  As far as I know the ergodox_ez code is exactly the same.  Did something change that should fix the issue?

drashna, 2018-10-22:
I've been going through stuff and trying to clean it up.  

I've been adding messages about re-opening, and didn't here. Sorry.

drashna, 2019-04-05:
Update your repo, and try this now.

IS_COMMAND has been changed, and now uses `get_mods`, which ignores the weak mods used by keycodes. (shifted keycodes).

So, this may no longer happen.

bendk, 2019-04-05:
It appears fixed to me.

drashna, 2019-04-05:
Awesome! 

Closing, in that case.

closed 2019-04-05 by drashna