qmk/qmk_firmware #2734: XD-002, a cool, new, but attiny85 2% keyboard

docteurzoidberg, 2018-04-12:
I know what you all right think when you are currently reading this,

"Oh no, not the tiny !" 
"we can't do anything on that uc !"
"so few space"

But I did a bet with a friend qmk can run it, but i wans't able to compile either the timer.c and bootloader.c files with the V-USB stack at the current state of the QMK repository.

But TMK seems to compile actually for attin85, without a lot of functions of course.

Is anyone able to guide be with re-enabling QMK on attiny85 so i can't try to add this nice good looking 2 key keyboard to the repo ?

Many thanks for reading and advices !

drashna, 2018-04-12:
I think the problem here is that the th ATtiny85 is limited to 8KB of program space (assuming no bootloader).

The problem is that there is just too much in QMK that it won't fit.  Even testing a 6 key keyboard and disabling just about EVERYTHING, it was larger than 8KB. 

This is including tricks such as enabling linker optimization, which can greatly reduce the size.  The smallest I could get it was 8.2K.

docteurzoidberg, 2018-04-15:
thanks for your answer ! 

I will try to see what can be removed, but seems more logic to keep the old tmk for this keyboard actually. 

Maybe a custom replacement PCB design for running QMK would be better

drashna, 2018-05-11:
Progress?

xtaran, 2018-08-24:
@docteurzoidberg: Any pointer to TMK for the XD002? So far I just found TMK for single key keyboards with the ATtiny85.

jackhumbert, 2020-03-17:
I'm gonna close this for now - if you have any updates, feel free to share them here!

PR #8598 (2020-05-01): Add xd002 support

## Description

Now with all the prerequisites merged, we can finally add some form of support.

### Future TODOs
- [ ] core bootloader support?
- [ ] save more space (eg bootloader code has "unused" func with eats 90 bytes) 

### Notes
With bootloader, 6.12k firmware space

Bare minimum
```console
Checking file size of xd002_default.hex                                                             [OK]
 * The firmware size is fine - 4014/6330 (63%, 2316 bytes free)
```

Not enough space for...lots of stuff
* rgblight `RGBLIGHT_ANIMATIONS`

As a general rule, you can enable _* [... cut]
closed 2020-03-17 by jackhumbert