qmk/qmk_firmware #2746: Auto shift trick

broken-username, 2018-04-14:
This isn't an issue so much as a suggestion. You might consider documenting that it's often helpful when using auto shift to, at least for certain words, develop the habit of holding the key down for just the tiniest amount of time extra, and immediately striking the next key in the word at the usual speed (still holding the first one down), then letting both go and continuing with the word or the rest of the sentence. The word won't print to the screen until after the second letter, and the first letter will be capitalized as you want. I get very consistent results this way. Hopefully that was explained well enough.

broken-username, 2018-04-15:
It certainly could be. I haven't played around with those settings personally (that I remember, anyway), so I would be interested in seeing how the effect might be reproduced. Let me know if you need more information, though this is the "Space Oddity" keyboard in the handwired folder for reference.

drashna, 2018-05-11:
IIRC, this is a tap/hold issue that affects a number of other keycodes/functions.

drashna, 2018-10-22:
Was this still an issue? 

Also, have you checked any of these settings? 
https://docs.qmk.fm/#/feature_advanced_keycodes?id=permissive-hold

n-scope, 2019-03-03:
@broken-username The behaviour you describe is exactly what I am trying to achieve. It enables super-fast autoshift. But too bad, I can't reproduce this behaviour.  (I have played around with "permissive hold" and interrupt mod tap" settings without success )

My main issue is that the first letter is released as soon as I press the second one. How can we achieve "fast autoshift" ?

IsaacElenbaas, 2020-05-08:
This is not an unintentional feature, it is simply how AutoShift works. When a key is pressed the previous is flushed if it wasn't released yet. You are holding the first key for longer than the tapping term, and the second clears it - this ensures keys are sent in the 'correct' order (as noone learned to type on AutoShift and releasing in the correct order would be horrible). You do save time by not waiting to release the first key to press the second, though.

fauxpark, 2020-11-29:
Closing, apparently #9826 has resolved this.

closed 2020-11-29 by fauxpark