qmk/qmk_firmware #2759: bmini, bfake do not build

engicoder, 2018-04-18:
Encountered errors trying to build the ps2avrGB based builds such as bmini and bfake from root of qmk.
Environment: Windows 10 with ubuntu WSL
Make command:  

> make bmini

Note: I am able to build other keyboards with out problem.

Errors returned:

```
Compiling: keyboards/bmini/matrix.c                                                                In file included from keyboards/bmini/matrix.c:18:0:
/usr/lib/avr/include/avr/io.h:428:6: error: #warning "device type not defined" [-Werror=cpp]
 #    warning "device type not defined"
      ^
keyboards/bmini/matrix.c: In function ‘matrix_init’:
keyboards/bmini/matrix.c:34:5: error: ‘DDRB’ undeclared (first use in this function)
     DDRB = 0xFF;
     ^
keyboards/bmini/matrix.c:34:5: note: each undeclared identifier is reported only once for each function it appears in
keyboards/bmini/matrix.c:35:5: error: ‘PORTB’ undeclared (first use in this function)
     PORTB = 0xFF;
     ^
keyboards/bmini/matrix.c:37:5: error: ‘DDRA’ undeclared (first use in this function)
     DDRA = 0x00;
     ^
keyboards/bmini/matrix.c:38:5: error: ‘DDRC’ undeclared (first use in this function)
     DDRC &= ~(0x111111<<2);
     ^
keyboards/bmini/matrix.c:39:5: error: ‘DDRD’ undeclared (first use in this function)
     DDRD &= ~(1<<PIND7);
     ^
keyboards/bmini/matrix.c:39:18: error: ‘PIND7’ undeclared (first use in this function)
     DDRD &= ~(1<<PIND7);
                  ^
keyboards/bmini/matrix.c:41:5: error: ‘PORTA’ undeclared (first use in this function)
     PORTA = 0xFF;
     ^
keyboards/bmini/matrix.c:42:5: error: ‘PORTC’ undeclared (first use in this function)
     PORTC |= (0b111111<<2);
     ^
keyboards/bmini/matrix.c:43:5: error: ‘PORTD’ undeclared (first use in this function)
     PORTD |= (1<<PIND7);
     ^
keyboards/bmini/matrix.c: In function ‘matrix_set_row_status’:
keyboards/bmini/matrix.c:53:5: error: ‘DDRB’ undeclared (first use in this function)
     DDRB = (1 << row);
     ^
keyboards/bmini/matrix.c:54:5: error: ‘PORTB’ undeclared (first use in this function)
     PORTB = ~(1 << row);
     ^
keyboards/bmini/matrix.c: In function ‘matrix_scan’:
keyboards/bmini/matrix.c:71:15: error: ‘PINA’ undeclared (first use in this function)
             (~PINA) & 0xFF
               ^
keyboards/bmini/matrix.c:74:27: error: ‘PINC’ undeclared (first use in this function)
             bit_reverse((~PINC) & 0xFF) << 8
                           ^
keyboards/bmini/matrix.c:77:16: error: ‘PIND’ undeclared (first use in this function)
             ((~PIND) & (1 << PIND7)) << 7
                ^
keyboards/bmini/matrix.c:77:30: error: ‘PIND7’ undeclared (first use in this function)
             ((~PIND) & (1 << PIND7)) << 7
                              ^
cc1: all warnings being treated as errors
```

drashna, 2018-04-18:
Odd.  it compiles fine for me.

Try merging/rebasing, and re-run the installation script.  Make sure the avr-gcc stuff gets downloaded and installed

engicoder, 2018-04-19:
Switch to linux (Ubuntu 16.04) and it does build fine as you said. Must be something specific to WSL.

closed 2018-04-19 by engicoder