qmk/qmk_firmware #2782: NOMOD mods modifier to clear all mods and shift layers?

sandric, 2018-04-21:
I love strict and simple way that `LCTL(kc)` or `RALT(kc)` allow me to add modifiers right in my keymap definition. By I struggle to find option to do reverse - simple delete all modifiers that is applied to keycode for example from layer, by using `LM(layer, mod)` as example. Right now, if you make new layer be activated with lets say `LM(1, LCTL)` - ALL of the defined keys in that mod will have left control modifier, but if you want only one keystroke on that layer to say print KC_A - there's no way to get rid of that control modifier by applying smth like `NOMOD(KC_A)`, right? Because of that all you can do is to either not use `LM(layer, mod)`, just ordinary `MO`, and wrap all keykodes on that layer with `LCTL(kc)`, leaving one `KC_A` out. Other way is macros. But I would prefer `NOMD` wrapper to keystroke to simply ignore any modifier - its much cleaner and understandable.

drashna, 2018-05-11:
Ah, for the LM feature.  

Maybe add a macro with `clear_mods();` in it?   Otherwise, I'm not sure.

jackhumbert, 2020-03-17:
@sandric were you able to implement something along these lines? A custom keycode like that is probably the quickest way to handle that.

sandric, 2020-03-17:
@jackhumbert If I did I have no recollection already unfortunately 😄 . Sorry about that.

jackhumbert, 2020-03-18:
Haha, fair enough! I'll close this for now then :)

closed 2020-03-18 by jackhumbert