qmk/qmk_firmware #2787: One Shot Mod cancelling

drashna, 2018-04-21:
Right now, there isn't a way to do this.  

However, it may be a good idea to add this.  So that when you hit the button again, outside of the tapping term, that it will cancel the OSM.

manogna4, 2018-04-26:
I thought "clear_oneshot_mods();" did this..Are you going to create a key code for it instead?

drashna, 2018-05-11:
But what happens when you don't want to press a key?  Especially if you have no timeout specified for it.

manogna4, 2018-05-12:
i don't understand your point about timeout. How would you go about cancelling a one shot mod without a timeout or pressing a key?

Also, I don't think it is a good strategy to use the same key (pressed just once) to cancel the one shot mod. That way (if you have timeout enabled), you will have to keep track of time to know if you are cancelling it or activating it again. If you don't have a timeout you will still not know if the mod is active or not. This happens with me with Caps Lock all the time. The price of that happening with 'Control' can be high.

drashna, 2018-05-12:
Right now, there is no way to cancel out of the modifier. So if you hit it, you're stuck with it until it times out, or until you hit a key.   

Personally, I do have indicators for this, so it's not a huge problem (I use the ergodox top LEDs for this, and have partially working code for RGB). 

But the issues is that sometimes, I'll hit it and then immediately change my mind.  The only solution is to wait for it to timeout, or to hit a key, backspace and then go back to typing.  It's not ideal, IMO.

mtourne, 2019-03-20:
I have the code below that seems to do the trick in most cases for canceling your OSM

it cancels the timeout and the stuck mods in most cases:
  * OSMLCTL + ESC -> cancels LCTL
  * OSMLCTL tap twice (stuck) + ESC -> unsticks LCTL

However I think there is a subtle bug with OSM:

  *  OSMLCTL (tapped) + OSMLSFT (held) : OSMLCTL is now stuck and get_oneshot_mods() and get_oneshot_locked_mods() both return nothing. yet I can see in my keyboard vizualizer than LCTL is still held.
  * OSMLCTL (tapped)  + OSMLSFT (tapped): works as expected 

```bool cancel_all_oneshots(void) {
  bool queue = true;

  uprintf("before: oneshot_mods(): %d, oneshot_locked_mods: %d, mods: %d\n", get_oneshot_mods(), get_oneshot_locked_mods(), get_mods());

  if (get_oneshot_mods() && !has_oneshot_mods_timed_out()) {
    clear_oneshot_mods();
    queue = false;
  }

  if (get_oneshot_locked_mods()) {
    clear_oneshot_mods();
    clear_oneshot_locked_mods();
    unregister_mods(get_mods());
    queue = false;
  }

  uprintf("after: oneshot_mods(): %d, oneshot_locked_mods: %d, mods: %d\n", get_oneshot_mods(), get_oneshot_locked_mods(), get_mods());

  return queue;
}
// Custom keycodes
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
  bool queue = true;

  // Cancel one-shot mods.
  switch (keycode) {
    case KC_ESC:
      if (record->event.pressed) {
        queue = cancel_all_oneshots();
      }
      break;

     [...]
  }

  return queue;
}

```

loeschzwerg, 2019-05-10:
You can tap the OSM key again, so that it becomes active as a hold modifier.
I guess this overrides the OSM state.

markstos, 2021-06-28:
A workaround to clear the one-shot mod state is to  use the `ONESHOT_TAP_TOGGLE` feature. 

For example, if this value is set to 3, quickly tapping the key 4 times will clear it. (The first three cause it to be held, but putting it in a state where the forth tap will clear the hold).

markstos, 2021-06-28:
The workaround of last resort is to unplug the keyboard and plug it back in 😂 .

azzamsa, 2021-09-07:
I use OSM for all the modifiers and I am always stuck in this state. Where I can't press any command/letter just because some modifier is still active. 

My current solutions are 1) using the Screenkey to show active/pressed keys 2) triple-tap the active OSM to disable it 3) if both don't work, I just plug and unplug. 

It is frustrating, but the strange thing is: I can't reproduce the situation. I tried to press many OSM consciously, but the timeout works.

However, in other times, the incident always happened. I don't know why the timeout doesn't work during those time. 

Any cleaner solution?

closed 2022-06-07 by drashna

luc-languagetools, 2022-12-17:
I got myself confused, it turns out my `OSM(MOD_LSFT)` does cancel itself when I double tap. The reason I thought it didn't is because I had shift double-tap set to enable caps word. Now, I have left+right shift for caps word, and double tap on my one shot shift turns off the shift. The double tap can even get me out of caps word, so i'm very happy now. Also configured the backlight to show when the one shots are active, no more confusion. QMK is really awesome !!