qmk/qmk_firmware #2796: chaining one shot mods

manogna4, 2018-04-24:
I would like to tap 'ctrl', tap 'shift' and then tap 'a' to get 'Ctrl+Shift+a'. At present this can be done with only one modifier. Tapping the second modifier cancels the first one. The above combination produces 'Shift+a' i.e. 'A'

This is a duplicate of [856](https://github.com/qmk/qmk_firmware/issues/856) and of [1580](https://github.com/qmk/qmk_firmware/issues/1580) but has been opened because @algernon asked for it on reddit/olkb. Both issues were closed because they were too difficult to do.

manogna4, 2018-04-24:
it works 😃 This is brilliant. Thanks.
Such a simple change too. I wonder why the the comments on the previous issues said it was too hard...

I'll let you know if it leads to any problems. I ran some simple tests, and it didn't cause any issues. I do not use any tapping toggles so I won't be able to tell you if it works with them.

algernon, 2018-04-24:
It may look simple, but it took me about four hours to find the right places to modify =)
It is almost always simple in hindsight.

manogna4, 2018-04-24:
whoa! Thanks a lot for your effort then. This frees up several keys on my layer.

drashna, 2018-04-24:
Awesome!

DanielGGordon, 2018-05-11:
Woah cool - can we make a PR for this?

algernon, 2018-05-11:
There is a PR for this, #2799.

PR #2799 (2018-05-14): Allow one-shot modifiers to be chained
Many a times one would want to use multiple modifiers with the same key, preferably without having to hold anything, like `Ctrl+Shift+C` or `Ctrl+Shift+V` to copy/paste in GNOME Terminal. To make this possible, we need to be able to chain one-shot modifiers, so that we can have multiple of them active at the same time.

The easiest way to accomplish this is that whenever we activate a one-shot modifier, we apply it on top of the existing set, instead of re-setting the state. When deactivating, either due to an interrupt, or due to a timeout, we deactivate [... cut]
commit 38f204db (2018-05-14): Allow one-shot modifiers to be chained

Many a times one would want to use multiple modifiers with the same key,
preferably without having to hold anything, like `Ctrl+Shift+C` or
`Ctrl+Shift+V` to copy/paste in GNOME Terminal. To make this possible, we need
to be able to chain one-shot modifiers, so that we can have multiple of them
active at the same time.

The easiest way to accomplish this is that whenever we activate a one-shot
modifier, we apply it on top of the existing set, instead of re-setting the
state. When deactivating, either due to an interrupt, or due to a timeout, we
deactivat [... cut]
closed 2018-05-14 by jackhumbert