qmk/qmk_firmware #2808: UT47 outputting incorrectly using the default keymap

Cpeters1982, 2018-04-25:
Alphas work as intended. Spacebar doesn't register. Lower layer (layer 2) numrow does not output the correct symbols as defined in the keymap.

drashna, 2018-04-25:
There was a fix for this, in theory. 

Do you have commit `4ad37331d31d7f4e734e51f2b38fde9732c47d98`?

drashna, 2018-04-26:
In `/keyboards/ut47/ut47.h`, if you change the `#define LAYOUT` to this: 
```c
#define LAYOUT( \
  K00, K01, K02, K03, K04, K05, K06, K07, K08, K09, K0a, K0b, \
  K10, K11, K12, K13, K14, K15, K16, K17, K18, K19, K1a, K1b, \
  K20, K21, K22, K23, K24, K25, K26, K27, K28, K29, K2a, K2b, \
  K30, K31, K32, K33, K34, K35, K36, K37, K38, K39, K3a, K3b  \
) \
	{ \
    { K00, K01, K02, K03, K04, K05, K06, K07, K08, K09, K0a, K0b }, \
    { K10, K11, K12, K13, K14, K15, K16, K17, K18, K19, K1a, K1b }, \
    { K20, K21, K22, K23, K24, K25, K26, K27, K28, K29, K2a, K2b }, \
    { K30, K31, K32, K33, K34, K35, K36, K37, K38, K39, K3a, K3b }  \
	}
```

Does this work?

Cpeters1982, 2018-04-26:
The above throws errors:

```
Compiling: keyboards/ut47/keymaps/default/keymap.c                                                 keyboards/ut47/keymaps/default/keymap.c:37:1: error: macro "LAYOUT" requires 48 arguments, but only 47 given
 ),
 ^
keyboards/ut47/keymaps/default/keymap.c:32:1: error: 'LAYOUT' undeclared here (not in a function); did you mean 'KC_OUT'?
 LAYOUT(
 ^~~~~~
 KC_OUT
keyboards/ut47/keymaps/default/keymap.c:44:1: error: macro "LAYOUT" requires 48 arguments, but only 47 given
 ),
 ^
keyboards/ut47/keymaps/default/keymap.c:51:1: error: macro "LAYOUT" requires 48 arguments, but only 47 given
 ),
 ^
keyboards/ut47/keymaps/default/keymap.c:58:1: error: macro "LAYOUT" requires 48 arguments, but only 47 given
 ),
 ^
 [ERRORS]
 |
 |
 |
make[1]: *** [.build/obj_ut47_default/keyboards/ut47/keymaps/default/keymap.o] Error 1
make: *** [ut47:default:avrdude] Error 1
Make finished with errors
```

drashna, 2018-04-26:
Could you post what that define looked like before?

And let me know exactly which buttons are wrong, and where the offset it?

Cpeters1982, 2018-04-26:
```
#define LAYOUT( \
  K00, K01, K02, K03, K04, K05, K06, K07, K08, K09, K0a, K0b, \
  K10, K11, K12, K13, K14, K15, K16, K17, K18, K19, K1a, K1b, \
  K20, K21, K22, K23, K24, K25, K26, K27, K28, K29, K2a, K2b, \
  K30, K31, K32, K33, K34,    K35,   K37, K38, K39, K3a, K3b  \
) \
	{ \
    { K00, K01, K02, K03, K04, K05, K06, K07, K08, K09, K0a, K0b }, \
    { K10, K11, K12, K13, K14, K15, K16, K17, K18, K19, K1a, K1b }, \
    { K20, K21, K22, K23, K24, K25, K26, K27, K28, K29, K2a, K2b }, \
    { K30, K31, K32, K33, K34, KC_NO, K35, K37, K38, K39, K3a, K3b }  \
	}
```
Previous layout define.

When switched to "lower" layer (2 in the default keymap)

Q = .
W = nothing
E = LShift LCtrl 1  according to Karabiner event viewer
R = '
T = ]
Y = F21
U = application according to Karabiner event viewer
I = F23
O = numpad .
P = application according to Karabiner event viewer

A = a
S = s
D = d
F = f
G = g
H = h
J = F19
K = execute according to Karabiner event viewer
L = =
; = .
' = F17
Z = F1
X = F2
C = F3
V = F4
B = F5
N = F6
M = F7
, = F8
. = F9
/ = F10

bottom row seems to be working as expected, except space doesn't register at all on Event Viewer or Keyboardtester.com.

drashna, 2018-04-26:
Honestly, this looks like it's the correct output, mostly.  The only issues being where the `KC_FN#` keycodes are.  Everything else appear to be spot on for what should be happening: 
https://github.com/qmk/qmk_firmware/blob/master/keyboards/ut47/keymaps/default/keymap.c#L46-L51

And if the base layer is working 100%, as well... then the issue is definitely the FN# codes.

Cpeters1982, 2018-04-27:
It's a little hard to type with no spacebar...

drashna, 2018-04-27:
Nah, you don't need a space bar. :) 

I would say check the solder joints on it. 
Otherwise, try changing `K35` to `K36` in the layout and see if that works (both instances of it)

Cpeters1982, 2018-04-27:
Also, I just flashed my modified map and everything works as intended... Except for spacebar. I'll reflow/add solder. You can close this, I think.

(Changing from K35 to K36 didn't do anything. I guess I can blame my _picture perfect_ soldering job... Maybe.)

drashna, 2018-04-27:
Try this then: 

```c
#define LAYOUT( \
  K00, K01, K02, K03, K04, K05, K06, K07, K08, K09, K0a, K0b, \
  K10, K11, K12, K13, K14, K15, K16, K17, K18, K19, K1a, K1b, \
  K20, K21, K22, K23, K24, K25, K26, K27, K28, K29, K2a, K2b, \
  K30, K31, K32, K33, K34, K35, K36, K37, K38, K39, K3a, KC_NO \
  ) \
  { \
    { K00, K01, K02, K03, K04, K05, K06, K07, K08, K09, K0a, K0b }, \
    { K10, K11, K12, K13, K14, K15, K16, K17, K18, K19, K1a, K1b }, \
    { K20, K21, K22, K23, K24, K25, K26, K27, K28, K29, K2a, K2b }, \
    { K30, K31, K32, K33, K34, K35, K36, K37, K38, K39, K3a, KC_NO}  \
  }
```

Cpeters1982, 2018-04-27:
That worked, sort of...

I had to change my keymap and out a "KC_NO" on the end of the bottom row because I kept getting the "error: macro "LAYOUT" requires 48 arguments, but only 47 given" error kept occurring on compile. 

So, space works now, but the right side of the bottom row is "shifted" one space to the left. I have no access to the "raise" layer and the last key on that row is dead. I'm assuming that's "working as intended" but it doesn't really solve the underlying issue, right?

I also tried removing the "KC_NO" from both the keymap and the ut47.h file, but I have the same result.

I also tried shifting the whole LAYOUT define bottom row one over. So, K35 became K36, 3y became 37, and ending with K3b. No change to the bottom row... Everything works but the last key on the right is dead and I have no access to "raise".

drashna, 2018-04-27:
Sorry, the matrix stuff still ... confuses me sometimes. 

It should have been: 
```c 
#define LAYOUT( \
  K00, K01, K02, K03, K04, K05, K06, K07, K08, K09, K0a, K0b, \
  K10, K11, K12, K13, K14, K15, K16, K17, K18, K19, K1a, K1b, \
  K20, K21, K22, K23, K24, K25, K26, K27, K28, K29, K2a, K2b, \
  K30, K31, K32, K33, K34, K35, K36, K37, K38, K39, K3a \
  ) \
  { \
    { K00, K01, K02, K03, K04, K05, K06, K07, K08, K09, K0a, K0b }, \
    { K10, K11, K12, K13, K14, K15, K16, K17, K18, K19, K1a, K1b }, \
    { K20, K21, K22, K23, K24, K25, K26, K27, K28, K29, K2a, K2b }, \
    { K30, K31, K32, K33, K34, K35, K36, K37, K38, K39, K3a, KC_NO}  \
  }
```

But sounds like this may be correct: 

```c
#define LAYOUT( \
  K00, K01, K02, K03, K04, K05, K06, K07, K08, K09, K0a, K0b, \
  K10, K11, K12, K13, K14, K15, K16, K17, K18, K19, K1a, K1b, \
  K20, K21, K22, K23, K24, K25, K26, K27, K28, K29, K2a, K2b, \
  K30, K31, K32, K33, K34,    K35,   K37, K38, K39, K3a, K3b  \
) \
	{ \
    { K00, K01, K02, K03, K04, K05, K06, K07, K08, K09, K0a, K0b }, \
    { K10, K11, K12, K13, K14, K15, K16, K17, K18, K19, K1a, K1b }, \
    { K20, K21, K22, K23, K24, K25, K26, K27, K28, K29, K2a, K2b }, \
    { K30, K31, K32, K33, K34, K35, KC_NO, K37, K38, K39, K3a, K3b }  \
	}

Cpeters1982, 2018-04-27:
Drashna, you are the man. Thanks so much for your help. That fixed it.

closed 2018-04-27 by Cpeters1982

drashna, 2018-04-27:
Yeah, I just noticed too that K35 is in K36's spot.  That's why.  The matrix is wrong.  Will fix.