qmk/qmk_firmware #2812: Alternate between Audio and Backlight functions.

nariox, 2018-04-26:
Originally posted on /r/olkb.

I've just purchased a DZ60, but unfortunately the audio-capable PINS are all being used. I'm wondering if I could solder a piezo buzzer to pin B6 (backlight pin) and use it simply to play the "STARTUP_SONG" and then switch it back to backlight mode. From what I understand, that would involve calling `audio_init()` and then `audio_toggle` after the startup song and then issue a `backlight_init` to start normal backlight usage, but I'm not sure if there would be adverse effects to doing that or if there's a better way to do that.

I realize that whenever I switch the backlight mode, it would cause the piezo to make noises (since it'd still be connected), but during backlight use, I'm hoping the PWM frequency is low enough not to be properly reproduced by the piezo. I did some calculations (after looking through some quantum code), for a Clock of 16MHz, with the 1/8 prescaler on timer 1 and ISR1 set to 0xFFFF, the frequency should be about 30Hz. Is that right? If so, that's probably low enough that the piezo would not pick up. Alternatively, I could use B2 pin to connect to a transistor as a "speaker enable".

tl;dr: what is the PWM frequency for the backlight on ATMega32u4? And can I temporarily enable Audio-mode and then toggle to Backlight mode on a pin (B6)?

Thank you

drashna, 2018-10-22:
I don't think there is a good way to do this, unfortunately.

jackhumbert, 2020-03-17:
I think you could do this if you want to have the backlight only at one brightness (always on). The LED will definitely affect the sound coming out of the speaker, though. If you have another pin available, it might make sense to move the speaker/backlight to that one, depending on what port you want to use for which. There are a couple of ports we support for audio: https://docs.qmk.fm/#/feature_audio?id=audio

nariox, 2020-03-17:
At least for the DZ60, we have a problem. For timer 1, B5 and B7 are part of the matrix, B6 is used for the backlight.

For timer 2, C4 and C5 are not available at the package and C6 is also used in the matrix. There are a few GPIO pins on the IC, but no good place to solder them.

Suppose I were to also enable audio on B6, would it be as simple as wiring it up in parallel to the LEDs?

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc