qmk/qmk_firmware #2813: MT() with KC_CAPS requires double-tap to get CAPS to engage—is that expected?

iakobou, 2018-04-26:
I posted this on /r/olkb [here](https://www.reddit.com/r/olkb/comments/8eqk10/trying_out_mod_tap_unsuccessfully_help/) initially, but . . .

> I removed the RGB code to see if it was conflicting. Issue persists. It's MT(MOD_HYPR, KC_CAPS) specific. LT(2, KC_CAPS) is just fine. MT(MOD_MEH, KC_TAB) and MT(MOD_HYPR, KC_ENT) work just fine. Is it expected behavior with MT in use with Caps or is it a bug?

I also tried MT(MOD_MEH, KC_CAPS) and it's the same issue—CAPS won't engage without a quick double-tap. It disengages with a single tap though. Is this normal behavior for CAPS or is it a bug?

drashna, 2018-04-28:
Tested this, doesn't happen for me.  Very odd. 

What board is this? 

Also, try playing with these settings: 
`IGNORE_MOD_TAP_INTERRUPT`, `PERMISSIVE_HOLD`, and `PREVENT_STUCK_MODIFIERS`, as these defines may affect this.

iakobou, 2018-04-28:
KBD75. If it isn’t happening for you, it must be a bug and is not expected behavior?

drashna, 2018-04-28:
Tried it on two different boards, and "worked as expected", but I have some of the above defined and undefined, so that may affect this. 

Specifically: 
```c
#define IGNORE_MOD_TAP_INTERRUPT
#undef PERMISSIVE_HOLD
#undef PREVENT_STUCK_MODIFIERS
```

And the KBD75 uses the same controller as what mine do ... 

So, have you tried updating your repo?

markspanbroek, 2018-08-17:
I'm experiencing the same. @iakobou: are you using a Mac? Apparently OSX has a slightly longer delay on the caps lock key. That's probably why a quick tap won't register.

https://forums.macrumors.com/threads/caps-lock-delay-problem-any-new-progress.1063089/

iakobou, 2018-08-17:
Correct, I’m on macOS 10.13.6.

I’ve removed the Caps MT() because I couldn’t get the simple syntax to work and didn’t want to add any DEFINEs.

closed 2018-10-22 by drashna