qmk/qmk_firmware #2834: RGBW White

JulioBBL, 2018-04-28:
i haven't found anywhere explaining how to activate the white channel on a RGBW led, am i not looking in the right place or there is no implemented way?

drashna, 2018-04-28:
Depending on what you're trying to do, the RGB underglow code should handle this automatically, I think. 

https://github.com/qmk/qmk_firmware/blob/master/quantum/rgblight_list.h#L20

JulioBBL, 2018-05-01:
But it sends tre values, for R, G and B respectivelly, but no value for W, and that is why i am here asking

drashna, 2018-05-11:
Ah, yeah, that would be a problem. 

What strip/LEDs are you using?

JulioBBL, 2018-05-19:
I am still planning my build, got an order for some SK6812, they are the same as WS2812B

drashna, 2018-10-22:
Any updates to this?

Taudris, 2019-01-25:
QMK doesn't have any real support for the W in RGBW. The only thing it can do is send commands to RGBW LEDs and have the correct RGB color show up.

I got RGBW working on my personal keymap by putting `RGBLIGHT_CUSTOM_DRIVER = yes` in `rules.mk` and defining a custom `rgblight_set()` that creates a copy of `led` (global variable which stores current LED values), then does some math to "convert" RGB to RGBW (find min channel value, subtract from RGB, add to W). This way, white can be controlled/blended with RGB by QMK's saturation controls.

kraigh, 2019-06-08:
@Taudris would you be willing to share your custom `rgblight_set()`? Maybe even contribute it to QMK?

drashna, 2019-11-21:
Stalebot bad.

spidey3, 2020-05-20:
Here is an article listing a potential algorithm to use to convert HSV to RGBW:
https://blog.saikoled.com/post/44677718712/how-to-convert-from-hsi-to-rgb-white

Note - there really is no good way to go from RGB to RGBW...

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc