qmk/qmk_firmware #2844: Suggested Linux udev rules are unsafe?

etale-cohomology, 2018-04-29:
On Linux, to [compile and install](https://docs.qmk.fm/faq/build-compile-qmk#cant-program-on-linux) the firmware you can do something like

```
make preonic:default:dfu
```

This won't work as is, however, because you need special (ie. root?) permissions. This problem can be circumvented by using `sudo` to gain temporary root privileges:

```
sudo make preonic:default:dfu
```

The docs, however, suggest another way: [changing udev rules](https://docs.qmk.fm/faq/build-compile-qmk#linux-udev-rules).

I think this may be unsafe.

To be sure, after changing the `udev` rules I could flash the firmware using `make preonic:default:dfu` (without root privileges!). But since the input device now has `0666` permissions (anyone can read and write), then anyone can read/write to the device!

So now, eg., in [C](https://en.wikipedia.org/wiki/C_(programming_language)) you can `open()` the `/dev/input` file where your keyboard is attached to (eg. `int fd = open("/dev/input/event2")`) and use the resulting file descriptor `fd` to `read()` all the keyboard input (eg. `read(fd, event, sizeof(event))`, where `event` is a `struct input_event`) and inject input with `write()` (eg. `write(fd, event, sizeof(event))`). As a regular user!

Isn't this bad? Please correct me if I'm wrong.

nariox, 2018-05-04:
You could set them to `660` and add yourself to the `input`. For "safest", you could create an user that is part of the `input` group and `sudo -u memberofinputgroup make preonic:default:dfu`

The problem with `sudo` would be if somehow your scripts became compromised and it installed or did something funky with your computer. But it should be relatively safe. In any case, when you install qmk, you are already trusting they didn't include a keylogger or something similar on your keyboard firmware anyway.

The problem with the `udev` rules is that any user would have access to it at all times (even hidden users or a malicious program). Using `660` would limit that to members of the `input` group, but if you are a member, similar problems apply. The `sudo -u memberofinputgroup` solves this, and also doesn't grant `root` privileges, so it's relatively more safe. But it's more cumbersome.

fredizzimo, 2018-05-04:
This depends on what you consider unsafe. The `udev` rules only exposes the bootloader interfaces of the keyboard, so there's no way to install a software based keylogger there for example. 

Theoretically it could allow someone to write a bad firmware to the keyboard, but you need physical access to the keyboard to put it into bootloader mode first.

I would assume that people who care this much about security also understands the consequences of it, and wouldn't create the udev rules, and use alternatives like @nariox suggests. For the rest of us, it's just an additional convenience.

closed 2018-10-22 by drashna