qmk/qmk_firmware #2853: Organize keyboards by categories

ericywl, 2018-04-30:
I think that organizing the keyboard by categories like designer or layout would help make it easier for people to find the keyboard they want, seeing as QMK is expanding quite rapidly now. Things like `M60-A`, `M10-A` and `RAMA/M6-A` makes it confusing and harder to find.

I would not mind helping out with the categorization process but would certainly need some pointers.

Wilba6582, 2018-04-30:
M60-A isn't even in there yet, and will be under the `rama` folder.

M10-A predates my additions of M6-A and eventually M10-B, M60-A, U80-A etc. I thought it best to leave it there until I add M10-B and then refactor and move under the `rama` folder.

ericywl, 2018-04-30:
@Wilba6582 Understandable, I was just using that as an example. I think what you are doing is what I'm hoping for - grouping keyboards together by desginers.

drashna, 2018-10-22:
I think that this has been done for a number of boards already.

closed 2019-12-21 by stale[bot]