qmk/qmk_firmware #2862: How can I toggle my LED when I toggle layer?

YipCharles, 2018-04-30:
My keymap's "Fn" just be set as toggle layer rather than momentary jump to layer, then I try to toggle the user defined LED at the mean time. But I can't find it in code, I think it will be somewhere in process_action, and try to do it under "ACT_LAYER_TAP" but just failed. Anyone who know about this?

drashna, 2018-05-10:
I think you want something like this: https://qmk.github.io/qmk_firmware/#/custom_quantum_functions?id=layer-change-code

aelg, 2018-06-04:
@drashna I can't figure out how to get this working when switching default layers (using `DF(layer)`). I don't get the `layer_state_set_user` call when switching **default** layer. `default_layer_state_set_user` doesn't seem to be implemented (only `default_layer_state_set_kb` and if I understand things correctly that shouldn't be used in keymaps?). 

What I'm currently doing is
```c
uint32_t layer_state_set_user(uint32_t state) {
    layer = biton32(default_layer_state);
    // set leds
}
```
The behavior I'm getting is,
switching default layers -> no effect,
then tapping a `MO(layer)`-key -> the leds are updated.

drashna, 2018-06-04:
I do this, actually. 

```
uint32_t layer_state_set_user(uint32_t state) {
#ifdef RGBLIGHT_ENABLE
  uint8_t default_layer = eeconfig_read_default_layer();
  if (userspace_config.rgb_layer_change) {
    switch (biton32(state)) {
    case _MACROS:
      rgblight_sethsv_noeeprom_orange();
      userspace_config.is_overwatch ? rgblight_mode_noeeprom(17) : rgblight_mode_noeeprom(18);
      break;
    case _MEDIA:
      rgblight_sethsv_noeeprom_chartreuse();
      rgblight_mode_noeeprom(22);
      break;
    case _GAMEPAD:
      rgblight_sethsv_noeeprom_orange();
      rgblight_mode_noeeprom(17);
      break;
    case _DIABLO:
      rgblight_sethsv_noeeprom_red();
      rgblight_mode_noeeprom(5);
      break;
    case _RAISE:
      rgblight_sethsv_noeeprom_yellow();
      rgblight_mode_noeeprom(5);
      break;
    case _LOWER:
      rgblight_sethsv_noeeprom_orange();
      rgblight_mode_noeeprom(5);
      break;
    case _ADJUST:
      rgblight_sethsv_noeeprom_red();
      rgblight_mode_noeeprom(23);
      break;
    default: //  for any other layers, or the default layer
      if (default_layer & (1UL << _COLEMAK)) {
        rgblight_sethsv_noeeprom_magenta();
      } else if (default_layer & (1UL << _DVORAK)) {
        rgblight_sethsv_noeeprom_green();
      } else if (default_layer & (1UL << _WORKMAN)) {
        rgblight_sethsv_noeeprom_goldenrod();
      } else {
        rgblight_sethsv_noeeprom_cyan();
      }
      biton32(state) == _MODS ? rgblight_mode_noeeprom(2) : rgblight_mode_noeeprom(1); // if _MODS layer is on, then breath to denote it
      break;
    }
  }
#endif // RGBLIGHT_ENABLE
```

aelg, 2018-06-10:
Thanks for the reply.
I think I framed my actual question badly. The problem is that it seems like there's no call to `layer_state_set_user` when the default layer is changed (ie. a `DF()` mapped key is pressed). The update works, but not until a *temporary* layer-switch happens (ie. a `MO()`-key is pressed). So the question is how do I detect a changed default layer?

drashna, 2018-06-10:
Well, if you check the `default` section, it checks the default layer, actually.  And I can verify that this works, since I use it in my code.
   
Though, there is also `default_layer_state_set_kb` that does get called immediately, when you change the default layer. 

There isn't a USER function for this, but you should be fine calling the Keyboard version of it, since it's not used.

aelg, 2018-06-10:
Ok I'll use the `_kb` func. And yes the default works, I just wanted to get the update immediately.
Thanks!

drashna, 2018-10-22:
There is a `_user` function for this now.

closed 2018-10-22 by drashna

Morid1n, 2019-12-23:
> Googling brought me here. So for the Googlers, for the future reference, I toggle my backlight made of a single led light like this (using `backlight_set()` method):
> 
> ```
> // My layer names
> #define _MAIN 0
> #define _FUNC 1
> 
> uint32_t layer_state_set_user(uint32_t state) {
>   switch (biton32(state)) {
>     case _MAIN:
>       backlight_set(3); // The number depends on your configuration's BACKLIGHT_LEVELS definition at config.h
>       break;
>     case _FUNC:
>       backlight_set(0);
>       break;
>   }
>   return state;
> }
> ```

I’m currently trying to figure this out for an M60-A Seq2. I want to light up specific keys when toggling my layer 1 which has arrow keys and have no idea how to go about it. Any tips? I’m a complete n00b when it comes to anything more than setting up a keymap.